prusa3d/Prusa-Firmware-Buddy #2729: [BUG] Printbed leveling problem

Ferrugius, 2022-11-17:
**Printer type** - MINI+

**Printer firmware version** - 4.4.0-RC1

**Original or Custom firmware** - Original

**Optional upgrades** - Wifi module

**USB drive or USB/Octoprint**
 Prusalink

**Describe the bug**
 There is a problem with bed levelling - the back of the bed is set to low in auto levelling causing the printer to print in the air. 

Downgrading to 4.4.0-BETA2 fixed the problem.

It seems to be exactly the same problem as described in this post from 2 years ago:
https://www.reddit.com/r/prusa3d/comments/nf2mz6/yaxis_trouble_with_the_mini/

**How to reproduce**
 Print any first layer test that prints in all for corners of the printer.

**Expected behavior**
The print head will be printing significantly higher (possibly even above the bed) in back corners of the print bed than in the front corners.

**G-code**
Will attach later - posting from a phone

**Crash dump file**
 Not applicable.

**Video**
Not available - already downgraded to fix the problem.

murk-sy, 2022-11-20:
Just to make sure, have you tried the official squaring guide? See Y axis checking especially.
https://help.prusa3d.com/article/squaring-your-mini_158518

I believe the bed leveling method was changed a bit in this RC to allow for some other future functionality, and it's possible that it does not tolerate the bed being a bit too much out of wack. If anything, I could say the first layers are now more consistent for me, but it's difficult to say for sure.

Ferrugius, 2022-11-24:
Yes I've tried absolutely every hardware calibration method there is and
only after downgrading firmware was the issue resolved. Maybe it was only
problem with my printer, but I thought I should report it anyway.

Prusa-Support, 2022-11-24:
Hello, thank you for the report! All noted on 4.4.0 Beta solving the issue, so it must be something on the RC1 to further review from our side. I passed this on to our devs to look into this in more depth and better understand the cause. In case of more info from our side, we'll update on this thread. 

Alessandro Pantaleo
Prusa Research

Prusa-Support, 2023-01-27:
Last week [FW 4.4.1 (final)](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1) was released. It seems to solve some movement inconsistencies and possibly related calibration problems (similarly to issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2326).

Please test it and share feedback as this issue could be closed.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-02-18:
Due to the recent firmware improvements that solved most likely connected issues, it seems safe to close this issue as solved.
Feel free to share further feedback though.

Michele Moramarco
Prusa Research

closed 2023-02-18 by Prusa-Support