prusa3d/Prusa-Firmware-Buddy #2732: [BUG] Newest file is not always selected when inserting USB

derenma, 2022-11-18:
**Printer type** - [MINI]

**Printer firmware version** - 4.4.0-RC1

**Original or Custom firmware** - Original RC

**Optional upgrades** - Bondtech extruder

**USB drive or USB/Octoprint** - USB

**Describe the bug**

Most of the time, the newest file on the USB stick is not automatically selected when it is inserted. (It may select either the last one printed, or get "stuck" selecting the same one over and over.)

**How to reproduce**

Depending on random conditions: Write new file from Prusa Slicer to USB. Insert drive into Mini. When the drive is inserted, it may select the previous print, not the new one. (You may have to write several files to the USB drive from Prusa Slicer for this to be truly replicatable. ("Replicatable".... NEW WORD! YAY!)

**Expected behavior**

Newest file is auto-selected when USB drive is inserted.

**G-code**

G-code does not seem to matter.

**Crash dump file**

No failure or crash occurs.

**Video**

[TBD]

derenma, 2022-11-18:
If I had to speculate, it may be related to this issue: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2690

Eldenroot, 2022-11-18:
This never worked for me... always one and the same file is selected... annoying!

murk-sy, 2022-11-20:
Duplicate of #2331

While time may be involved, this issue appears even when timestamps all appear correct - and then a 10th or 4th newest file is presented for printing.

I believe as a workaround you can backup all files, format the drive, and put the files back on. It will work normally afterwards.

derenma, 2022-11-20:
@murk-sy Thanks for the dup note and the workaround! :)

closed 2022-11-20 by derenma