prusa3d/Prusa-Firmware-Buddy #2733: [BUG] Filament type changes after running out of filament

twshield, 2022-11-18:
**Printer type**
MINI+

**Printer firmware version** 
latest 4.3.4

**Original or Custom firmware** 
original

**Optional upgrades** 
Filament Runout Sensor

**USB drive or USB/Octoprint**
 USB flash drive

**Describe the bug**
Just used up my first spool of PETG filament.  Had an identical one to replace it with.  Filament sensor worked as expected (although it might be nicer to wait for user input before ejecting old filament).  On replacing filament with identical filament the displayed type (and type shown via web interface) changed from PETG to PLA.

**How to reproduce**
 
1. Print with PETG until it runs out.
2. Start printing again with filament replaced (there was never a prompt for filament type, which makes sense)
3. LCD display now shows PLA as filament type (there was not a nozzle temperature change).

**Expected behavior**
  Expected the filament type to stay as PETG.

**G-code**
  G-Code was from PrusaSlicer 2.4.2, but I don't believe this matters.

**Crash dump file**
  n/a

**Video**
  n/a

twshield, 2022-11-18:
One further note: After the print is done and you go to unload the filament (only way I know to reset the filament type), it uses the wrong nozzle temperature, which is not a problem with PLA instead of PETG, but might be for filaments that need a much higher nozzle temperature than PLA does.

murk-sy, 2022-11-20:
Duplicate of #1400

This is fixed in the latest available release candidate [4.4.0-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-RC1). Consider giving it a try.

twshield, 2022-11-20:
Thanks, I did search but did not find that for some reason.

closed 2022-11-22 by twshield