prusa3d/Prusa-Firmware-Buddy #2734: [BUG] Target-Temperature after filament load not shown correctly

Saij, 2022-11-19:
**Printer type** - Prusa MINI+

**Printer firmware version** - 4.4.0-RC1

**Original or Custom firmware** - Original

**Optional upgrades** - Bondtech IFS Extruder

**USB drive or USB/Octoprint** - USB

**Describe the bug**
After loading the filament, the nozzle is going to cooldown to 170°C (nozzle temperature shown in blue in the footer). But it always displays the target temperature used to load the filament instead of 170°C in the footer.
If you look at the temperature menu it shows correctly 170°C and if you set it it will also shown in the footer correctly.
Same happens when you select a preheat option. The heating stops at 170°C but if always shows the target temperature you selected for preheating.

**Expected behavior**
Display the currently set target temperature in the footer after changing filament or while preheating.

Eldenroot, 2022-11-19:
Nope, this is correct.

It is preheated for mesh leveling step which requires 170 °C.

murk-sy, 2022-11-20:
Duplicate of #2616, see comments there for details

Prusa-Support, 2022-11-24:
Hello! Thank you for the report and for the comments, I confirm this is regular behaviour, intended to limit the oozing of filament from nozzle during the Mesh Bed Leveling. Closing this specific issue for this reason. 

Alessandro Pantaleo
Prusa Research

closed 2022-11-24 by Prusa-Support