prusa3d/Prusa-Firmware-Buddy #2760: [ENHANCEMENT] Wait for temperature to stabilise after loading new filament after cooldown

murk-sy, 2022-11-25:
**Printer type** - MINI
**Optional upgrades** - Filament Runout Sensor

**Is your suggested improvement related to an existing problem? Please describe.**
A few times when changing filament - specifically PA on a print with a few hundred tiny pieces - after loading the filament immediately on temperature and continuing the print, the parts exhibited a bulge on the layer where the new layer was added, which made me a bit wary of not checking filament amount before printing.

However, next time it happened, after letting nozzle heat up and extrude a bit (and likely filament inside it too), before confirming filament extrusion, I waited for some additional time for the temperature to stabilise (maybe 30 seconds to a minute at most?), and after continuing, the parts exhibited no defects at all.

The parts printed were basically d8 x 20mm cylinders that change to about 6mm diameter halfway up, example 3mf below.

I honestly don't know if the issue is reproducible with other materials, it might be just a combination of small parts and PA filament.

**Describe the expected functionality**
1) Printer runs out of filament
2) Nozzle cools down to ambient after 30 minute timeout
3) User "wakes" printer and loads filament
5) Filament is auto-loaded and extruded as soon as temperature is reached
6) User confirms extrusion/color
7) **Printer waits an amount of time for temperature to stabilise**
8) Small amount is extruded (as usually) and the print continues

**Sample project**
[Wait for temp after cooldown.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/10089532/Wait.for.temp.after.cooldown.zip)

Eldenroot, 2022-11-25:
I agree, this happens.

Needs to by solved with some stabilization time and maybe user confirmation.

Prusa-Support, 2022-12-08:
Thanks for the description of the problem.
I'm afraid we are talking about regular oozing.
Once melted, the plastic will naturally flow down due to gravity, similarly to a liquid.

If the printer awaits for a certain period of time, before continuing the print, having the nozzle at the print temperature, a varying amount of plastic will ooze from the nozzle - sometimes enough to mess up with the print unless the user will properly check and clean the nozzle.

You can expect a bit more oozing on Bowden Extruders (like the MINI's one) in comparison with the Direct Drive Extruders (like the MK3's one), but there is little to improve on the firmware side.
Awaiting while the nozzle is hot may worsen things if the user won't clean the nozzle.
It may rather help if the extruder retracts a little, but that may cause a temporary clog (or under-extrusion) in case of excessive oozed plastic.

- Please wait for the filament to load, and you can wait a bit longer then.
- Before confirming that "the color is clear" to resume, wait several seconds and clean the nozzle.
- Observe a few more seconds and clean again if oozing is still considerable, then confirm to resume the print.

The same applies to simply paused prints, in fact, we already have an issue opened to possibly improve this behavior https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/264.

I hope this clarifies.
Can this issue be closed?

Michele Moramarco
Prusa Research

closed 2022-12-12 by murk-sy