prusa3d/Prusa-Firmware-Buddy #2767: [BUG] [4.4.0] Auto bed leveling failed, USB connection failed after 4.4.0 Firmware update for the Prusa MINI

xdagee, 2022-11-25:
**Printer type** - MINI

**Printer firmware version** - 4.4.0

**Original or Custom firmware** - Original Firmware

**USB drive or USB/Octoprint** - Wired USB Connection

**Describe the bug**
  Autobed leveling failed after updating to 4.4.0. After updating from 4.3.4 to 4.4.0, I did a factory rest and went through the inital set up wizard. All test were completed but when i got to the first layer calibration, it failed at auto bed leveling.
I went ahead with to my PC only to notice USB also not connecting with pronterface. 

**How to reproduce**
  Download the new update and flash via USB and do a factory rest and following up with the initail setup wizard.

**Expected behavior**
  I'm expecting auto bed leveling to be completly dine so i can do my first layer print and try out the new update.

murk-sy, 2022-11-27:
May be related to #2729, also ABL issues

Can you try downgrading to 4.4.0-BETA2 and see if it functions there?

xdagee, 2022-11-28:
> May be related to #2729, also ABL issues
> 
> Can you try downgrading to 4.4.0-BETA2 and see if it functions there?

Alright, I will do that and report now, downloading 4.4.0-BETA2.

murk-sy, 2022-11-29:
Extra question the devs will probably want to know, do you have the Mini or Mini+? The issue I linked appears to be Mini+, but just to make sure. To be clear if you ordered in October 2020 it's probably already the Mini+.

xdagee, 2022-11-29:
> Extra question the devs will probably want to know, do you have the Mini or Mini+? The issue I linked appears to be Mini+, but just to make sure. To be clear if you ordered in October 2020 it's probably already the Mini+.

Alright, mine is the original prusa mini

xdagee, 2022-11-30:
Got it fixed with version 4.4.0-BETA2

murk-sy, 2022-12-02:
You should reopen the bug, working in beta but not working in release means it's not fixed or that it was broken afterwards.

afonsosantos, 2022-12-10:
Same issue with the USB connection, following it on this issue: #2815

zoltan-l, 2022-12-11:
Does the MBL fail also when you run it from menu?

xdagee, 2022-12-11:
> Does the MBL fail also when you run it from menu?

Yes, MBL fails as well.

Prusa-Support, 2023-01-06:
Thanks for reporting.

The two issues mentioned in this thread may not be related and may deserve separate issues.

As mentioned in https://github.com/prusa3d/Original-Prusa-MINI/issues/42#issuecomment-1370583454, we are aware of USB connection issues that arose with the recent releases.
The USB connection on the last release [FW 4.4.1-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1-RC1) should be more reliable. Further improvements should come with the upcoming release (#2844).
A relevant issue dedicated to the USB connection can be followed here https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2815.

The calibration anomaly is probably related to https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2326 or https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2729. The first issue is solved by [FW 4.4.0-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-RC1) though, and the second one is manifested in a slightly different way.

Please test [FW 4.4.1-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1-RC1) if you don't mind.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-01-27:
Last week [FW 4.4.1 (final)](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1) was released. It fixes some serial connection problems which could be related to this issue.

Please provide feedback so that this issue could be possibly closed as solved.

Michele Moramarco
Prusa Research

xdagee, 2023-02-03:
### I'm sorry but it doesn't seems to work for me, on a fifth try and still same issue.

**steps to replicate the issue.**
1. Download, copy and flash the latest firmware update (FW 4.4.1)
2. After a successful install
3. Run a factory rest
4. Follow the onscreen wizard to process
5. When it get to first layer calibration
6. Auto bed leveling fails 

attached is a screenshot on the LCD screen
[image]

Prusa-Support, 2023-02-14:
Could you please confirm that the problem is gone after downgrading to an older firmware version?

Regardless of that, I don't want that this is a case of mixed luck due to an even slightly skewed geometry, at the edge of what the firmware can handle. Maybe the recent firmware modifications made it a tad more sensitive.

Please check the offset between nozzle and bed along the X/Y axes, and take pictures, similar to [this one](https://cdn.help.prusa3d.com/wp-content/uploads/2021/08/5d0018f555e2e04c_painted-800x533.jpeg).
Spin the Z lead screw by hand until the nozzle almost touches the bed.
Then slide the X and Y axis by hand to take pictures of the distance between the nozzle and bed, at the 4 corners of the bed from a flat angle.
You can take multiple pictures from different orientations at each corner.

Michele Moramarco
Prusa Research

xdagee, 2023-03-25:
yes please, I can confirm the problem is gone after downgrading the firmware. I downgraded to firmware 4.4.0-BETA2. 
Following the downgrading to firmware 4.4.0-BETA2, I have tried with 4.4.1, 4.4.1-RC2, 4.4.1-RC1 and still the same issue.

I will replicate the steps as instructed in the suggestion and I will share update soon. Thank you

Sorry for the late reply.

xdagee, 2023-04-04:
> yes please, I can confirm the problem is gone after downgrading the firmware. I downgraded to firmware 4.4.0-BETA2. Following the downgrading to firmware 4.4.0-BETA2, I have tried with 4.4.1, [...]

update
Here are the pictures
https://photos.app.goo.gl/MQteaZhcpVJkWH8C9

xdagee, 2023-04-05:
> update Here are the pictures https://photos.app.goo.gl/MQteaZhcpVJkWH8C9

new discoveries, link here - https://forum.prusa3d.com/forum/postid/649899/

Prusa-Support, 2023-04-15:
Thanks @xdagee. There is light behind the printer though.
Backlighting and shadows on the nozzle may not help.
Also, please take pictures from a lower angulation, so that the heatbed won't be visible.
A light color background surely helps but possibly avoid backlit pictures.
Here is an example (sorry for the messy nozzle).

[image]

Also, I'd like to make sure the nozzle is as close as possible to the sheet, and that you didn't move the axes "via settings" but pushed the parts by hand, otherwise the printer will move the Z axis dynamically along the X/Y axes.  The nozzle may drag on the sheet in case of slight skew so please stop if you see that the nozzle gets in contact with the sheet (in case of doubts, check if you can fit some paper below).

Michele Moramarco
Prusa Research

xdagee, 2024-05-27:
**This issue hasn't been fixed.**
The user became _frustrated_ with the back-and-forth messages, which seemed impossible to follow up on for a fix, and decided not to follow up again.

Prusa-Support, 2024-07-20:
I'm sorry to say that we are unable to reproduce and confirm the problem.
After more than one year, it seems safe to say this is not a bug or a problem on the firmware so I strongly recommend solving this problem with the direct help of our Customer Support and closing this issue, as probably no firmware upgrade can help it.

The problem is supposedly specific to a hardware or assembly problem and sadly, it may not find solutions in this repository.

Michele Moramarco
Prusa Research

xdagee, 2024-07-21:
agreed.

closed 2024-07-21 by xdagee