prusa3d/Prusa-Firmware-Buddy #2768: Stuck hotend detection misfunctioning

mix579, 2022-11-25:
I spent the afternoon replacing the thermistor because I thought it was broken but I'm getting the same error with the new thermistor. It's the "TEMP NOT MATCHING
 error described in the 4.4 release notes under "Detect stuck hotend temperature reading". However, I'm only getting the error when trying to load ASA, not with PLA or PETG. The main problem is that it then aborts the load process, and I'm stuck with filament loaded into the extruder. Can't unload it because same error. Work around is to manually raise the nozzle temperature, which sometimes also throws the error but eventually works. I don't think there's anything wrong with my printer, I've been printing ASA on it for a long time. 

Mini+, 4.4.0 -RC1 firmware, Dragon hotend.

thijstriemstra, 2022-11-26:
> Mini+, 4.4.0 -RC1 firmware

Latest release is 4.4.0, you might want to give that a try.

mix579, 2022-11-26:
Which of course I did. Same results.

Particularly bothersome when you finished printing, then want to eject the filament because it doesn't gracefully recover from the error and starts cooling down and there's nothing I can do short of rebooting and then manually heating the nozzle to target temperature before initiating the unload process. I haven't tried it but it feels like this makes color swaps basically impossible because I can't unload the original filament without interrupting the print with a reboot.

Prusa-Support, 2022-12-10:
Thanks for your feedback.

The feature was supposedly tested in the possibly worst scenarios (described in https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/1552) but there are known issues (described in the same Pull Request).
A known issue is that the printer freezes if the error is triggered during the preheat, so I guess the "main problem" of this issue could be called a "known issue".

.

About the error itself, we can't exclude that the error is triggered incorrectly/prematurely because "this safety feature is currently still in beta and might misbehave" ([releases notes v4.4.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0)), but it was tested quite thoroughly and reasonable "padding" was added to the thresholds to prevent fake triggering. There are chances that here we are looking at a hardware problem.

I can speculate that it may not be a coincidence that the problem only occurs at specifically high temperatures (ASA). Some unnoticeably small damages on the electronics (cables, boards...) may manifest as the matter expands due to an increase in temperature. In these cases, most of the time you can expect the problem to occur more frequently as the existing (yet unnoticeable) damage "grows" over time.

Our developers may want to look into this.
Meantime, please closely look for damages on the thermistor head and its cable, and the Buddy board itself. Also double-check the assembly and the components around: a not so well centered thermistor, the thermistor head compressed by the set screw, fans incorrectly aiming or spinning a bit of the range...
Also consider trying other (possibly Prusa genuine) thermistors and downgrading firmware to closely observe the temperatures while operating without fake errors (stuck temperature below the target, temperature overshooting...).

Michele Moramarco
Prusa Research

dsmccusk, 2022-12-11:
I checked all my connections and temperatures and couldn't come up with a logical solution, so I downgraded the firmware back to 4.3.4 and it's working well.  I'll monitor the temps for a while, but at least for today they seem consistently on target.

mix579, 2022-12-11:
Same here. Don't see anything wrong with the connections. When I replaced the thermistor, I used a genuine Prusa part. I have a hard time thinking it's thermistor-related. As a workaround I heat the nozzle up to 170 degrees, and then from there I go through the regular loading procedure. I don't get errors if I do it this way,  and I'd think if it were thermistor-related I'd see the error as well. Maybe the algorithm used doesn't work well with a Dragon hotend even though I've never seen any thermal differences between it and the stock hotend. My preferred solution—short of some change to the algorithm—would be an option (maybe in the hidden hardware menu) to disable the check.

Prusa-Support, 2022-12-21:
Please give it a try with Prusa genuine parts (hotend, fans, design...) if you don't mind, and let us know if the issue is confirmed with the use of specific third-party components.

Michele Moramarco
Prusa Research

avancade, 2023-01-22:
I received my Prusa 3D Mini+ printer today and updated the firmware to 4.4.1 and I'm experiencing the same issue with. Of course these are all genuine parts.

mix579, 2023-01-22:
Please, for the love of god, give us an option to disable this temp check, like it's possible with the new Mk3S thermal protection function.

parabolala, 2023-01-22:
I have received the new printer and have got the same alert. Disconnecting the temp sensor from the board and reconnecting it resolved the problem for me.

dsmccusk, 2023-01-22:
I think I was able to correct this finally as well.  I broke it down and removed and inspected the hotend thermistor more closely and it looked like the wires weren't seating well enough into the connector.  I pushed in the wires as far as they would go and gave it a little squeeze, plugged it back in, ensuring there was no tension on the wire entering the connector.  Have not seen the issue again since.

your-friend-alice, 2023-02-02:
The issue is really bad with the Dragon hotend. Please make this feature optional, this seriously hampers the ability to mod this printer, which especially stings when the stock hotend has essentially fatal heat creep issues.

Lukexxr, 2023-02-07:
Had the same issue with brand new prusa mini+ with 4.4.1. Downgraded do 4.3.4 and the issue is gone.

Prusa-Support, 2023-02-08:
The problem may be concerning the faulty or misaligned thermistor.

In the case of genuine Prusa hardware, I suggest reaching our Customer Support for an evaluation of the problem.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research

DoNovae, 2023-03-11:
I have got the same problem with ASA.
Info : 
- Original Prusa MINI+ Semi-assembled 3D 
- Firmware 4.4.1 or  4.3.4

DoNovae, 2023-03-23:
...
I solved it with the help of support, by just unplugging and replugging the Molex connector of the nozzle.

tomelm, 2023-04-13:
Hi! I've been getting this error recently after installing the Slice Engineering Mosquito hotend. I previously used the Copperhead hotend successfully with no issue. I've done the following testing since I also recently got brand new heater + thermistor from Prusa and Slice.

| | Copperhead      | Mosquito |
| - | ----------- | ----------- |
| Prusa heater + Prusa thermistor | ✅ works, no error      | ❌ does not work, error       |
| Prusa heater + Slice thermistor | ❌ does not work, error   | ❌ does not work, error        |
| Slice heater + Slice thermistor | ❌ does not work, error   | ❌ does not work, error        |

For the test with the Prusa heater + thermistor: I did not unplug the cables from the board, I only swapped in/out the heater and thermistor.

Prusa-Support, 2023-04-15:
@tomelm repositioning the heater and thermistor may have helped. Not sure about the tests with third-party components, the result would be difficult to predict because we mainly work with genuine Prusa hardware.

.

So far I may seem to understand that most of the users in the thread, and customers reaching out to our support, solved the problem by
- reseating the thermistor connector,
- realigning (flush) thermistor and heater capsules in the heater-block,
- reverting to genuine Prusa parts (if third-party HW was used),
- performing a factory reset + new FW installation (in rare firmware corruption cases),  
- with the help of Prusa Customer Support (we can evaluate if a replacement is due in the case of genuine Prusa parts).

The result is hardly predictable with third-party components.
The developers may come out with alternative solutions though.
We are still investigating.

Michele Moramarco
Prusa Research

your-friend-alice, 2023-04-15:
one possible solution to preserve the tighter safety controls for unpredictable third-party hardware: have a calibration routine the customer can run to tune the safety parameters to their own setup. like cycle the hotend through a few different setpoints, measure the ramp rates (or whatever other metrics the safety system measures), and save those to EEPROM as the new "expected" values to compare  against. it's a compromise on safety since you're now comparing the printer's performance against a customer-supplied definition of nominal, instead of one that's been fully validated by Prusa, but i think if you bury it in the advanced menu and add a warning about these caveats it would be a good balance

Prusa-Support, 2023-05-09:
The current temperature monitoring system is flexible and resilient enough to work in varying scenarios and edge cases. This is surely a good point though.
Thanks for your suggestion.

Michele Moramarco
Prusa Research

mix579, 2023-05-09:
I support @your-friend-alice's suggestion for some mechanism to create a customized profile, or much easier, just an option to disable the thermal profile. 

At this point you're basically bricking customer's printers with perfectly legit mods such as a Revo or a Dragon that used to work perfectly fine. The only remedy is to go back to an old firmware version. Now, if you have shipped older firmware and deemed it safe to do so, it stands to reason that the new thermal protection is an additional safeguard (for which I applaud Prusa) but not required for safe operation. In which case nothing stands in the way of adding a switch to turn it off.

your-friend-alice, 2023-05-10:
to be clear, my above comment is not a bare minimum, it's a best case scenario. i don't think prusa needs to do what i suggested, but they do need to do *something*. The firmware probably works fine for the majority of users, but i don't trust that majority to stay large enough over time to match up with the reputation I keep hearing about prusa from their other printer models.

it's a shame the PCB has that breakaway firmware locking tab that kind of acts as a big billboard that community development is not welcome; this seems like the sort of problem that a robust community fork could hammer out a solution for pretty quick if it existed, while if a prusa dev wanted to go to bat for this, they'd be fighting a losing battle trying to convince management it's a worthwhile investment of effort.

Prusa-Support, 2023-06-08:
Thank you again for your input.

Now more than ever before, if an additional measure in rare cases can salvage several meters of filament, or extend the hardware lifespan by a few months, we can't really ignore that.
We can't underestimate safety but your opinion matters, and we are taking notes.
Our developers will surely take your suggestion into account and make their considerations in accordance with the company's legit goals.

I'm afraid this is slightly slipping out of the topic but the [appendix](https://help.prusa3d.com/article/flashing-custom-firmware-mini_14) is there to protect inexperienced users from uncertified firmware versions that may promulgate in the wild internet.
Experienced users may consider breaking it at their own risk.
Also, as you probably already know, the warranty time is limited, so there are virtually no consequences for breaking the appendix once the warranty is over (it is over for a large majority of printers out there). 
We recommend caution when tweaking the firmware or flashing unsigned versions - especially as long as modding is not strictly necessary but implies a necessary even slight margin of risk. Nonetheless, the whole community and compatible parts producers can freely release a custom firmware altogether with their third-party components.

Michele Moramarco
Prusa Research

ehariton, 2023-07-06:
Had this issue with my Prusa Mini (100% genuine parts) pre-upgrading to 4.4. Temp on the fillament heater was unable to get past 212 degrees and fillament was stuck half-way loaded. Updated firmware to 4.4, error persisted and showed "measured temp is not matching expected value error". Self-test hot-end heater failed.

Solution:
1) Manually loaded fillament
2) checked thermister (13.8ohm) and hot-end heater (73kohm) resistances: https://help.prusa3d.com/article/multimeter-usage_2117

After plugging thermister and hot-end heater cables back-in, self-test passed and printing again.

Prusa-Support, 2023-09-27:
Among other causes of failing thermistors, this may definitely be the case of a short thermistor cable as the cable may be tensed during the wiring finalization and pulled over the movements.

Please make sure the thermistor cable doesn't appear to be short in the electronics box and that the connector is not loose.
You may also want to check if the connector and port wobble on the board as the connector may be damaged or pulled together with the socket.
To be on the safe side, please check the heater cable and connector in the same way.

So far my best advice would be to consult the Customer Support before posting further reports.

Michele Moramarco
Prusa Research

bobc, 2023-12-22:
I have the same problem, used to work fine, upgrade to 5.1.2 kills my perfectly functioning printer.

This is really the last straw, I'm never buying Prusa again.

Prusa-Support, 2024-01-25:
We respect your opinion, however, so far, the cause of the issue apparently was the use of third-party components, or malfunctioning hardware - which is the ultimate reason for this safety feature to exist.

In the latter case, our Support should be able to address the problem.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research

ehariton, 2024-05-28:
Follow-up. I kept having these errors till I replaced the original thermister that came with the unit with another one also ordered from Prusa. Software wasn't the issue.

closed 2024-08-27 by github-actions[bot]