prusa3d/Prusa-Firmware-Buddy #2769: [BUG] Mini keep asking to flash ESP

BastR, 2022-11-26:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - [4.4.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint**
USB Drive

**Describe the bug**
The mini was in 4.4.0 beta for several weeks and worked perfectly. I had installed an ESP from AZDelivery and it worked great.

I upgraded to the final version and the mini asked me to flash the ESP again but it didn't work, I never got past step 1. The wifi doesn't work anymore either.

I changed the ESP with another one from the same supplier, the mini managed to flash the ESP and then connected to the wifi BUT when I restarted the printer, the mini asked me to flash the ESP again but this time it didn't work...

mix579, 2022-11-26:
Yes, it seems to be finicky. I was doing maintenance and updated from RC1 to final version, no problems including flashing the ESP. I then moved the printer back into its enclosure, and realized I had forgotten to re-create the .ini file with the wifi information. Did that, printer refused to load it. Tried to reflash the ESP, didn't work. Restarted, reflash worked fine and the printer loaded the .ini file no problem.

tedmahsun, 2022-11-27:
Ah, glad I'm not the only one having this problem. My symptoms are the same as BastR. Worked fine during Beta and RC1, but when I upgraded to final, it keeps asking me to update the ESP firmware every time I boot up, and then fails if I confirm. Printer continues to work fine though. It's just annoying having to click through those error screens on boot every single time.

bmorgenthaler, 2022-11-28:
I too have this issue, I've tried rebooting, pulling power, I've verified the .ini file is in place, etc. Still comes up every reboot. I don't know if the wireless works because honestly I'm using Ethernet and just wanted to have the ESP in it to test the beta originally.

tr-v-r, 2022-12-02:
I ran into this issue as well. I was running RC1 prior to updating to the final version. For me the repeated prompts to reflash the ESP started after I used the reset button. The flash attempt repeatedly failed.

The following procedure worked for me and wifi is up and running again, I'd be curious if others can replicate this behavior.

- I ensured wifi was toggled off in Settings > Network > Interface 
- Restarted the device using the power switch
- When the printer came back up, I reconfigured the Wifi in Settings > Network > Wifi > Setup Wi-Fi module
- The flash was successful

It's been a few days now and I had not had the wifi setup/flash prompt again.

After reading this report, I decided to try the reset button again. When the printer came up, I was back in the cycle of needing to setup wifi and the flash failing.

big-fudge, 2022-12-16:
Same here, but nothing on reboot. Fails to flash the ESP Module and 0/3 steps. Beta and RC1 were fine and without any problems.
Did all the steps from other posts, still same issue.

Prusa-Support, 2023-01-12:
Thank you all for your reports.

As highlighted in issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2312, there could be various reasons for the ESP Wi-Fi module not to be flashed correctly.

Here we are talking about the final FW 4.4.0 though, where a few problems with Wi-Fi connection and ESP flashing were found and already improved in the following few firmware releases.

The last firmware should solve and close this issue.
Possibly please test [FW 4.4.1-RC2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1-RC2). 

Michele Moramarco
Prusa Research

closed 2024-06-19 by github-actions[bot]