prusa3d/Prusa-Firmware-Buddy #2771: [BUG] dot-underscore version of GCODE created  & auto-selected

s0urfruit, 2022-11-26:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - 4.4.0 (and earlier, upon update to versions with revamped PrusaLink) 

**Original or Custom firmware** - [Original]

**Optional upgrades** - None

**USB drive or USB/Octoprint**
  USB drive

**Describe the bug**
Prusa MINI displays two files every time an STL is sliced, and GCODE is placed on a USB drive. One of the files is (e.g) "flowers.gcode" which prints fine, but the other is "._flowers.gcode" which "prints" instantly with a print time of 0 minutes & no model preview.

**How to reproduce**
- Slice any file using PrusaSlicer, (e.g) PS 2.4.0. 
- Place sliced file onto USB
- Insert USB into MINI
- Enter the USB drive on the MINI UI
- See two files

**Expected behavior**
I did not expect "._flowers.gcode", nor do I expect it to be auto-selected when inserted into MINI.

**G-code**
Irrelevant?

**Crash dump file**
None.

**Video**
Can attach in a comment, printer is currently printing 10hr model.

s0urfruit, 2022-11-26:
Here's a picture, sliced using PS 2.5.0. [image]

Eldenroot, 2022-11-28:
I am not the only one :)

Prusa-Support, 2022-12-01:
Hello, thank you for the report! We are aware of this, we are working to improve this aspect of hidden files being shown on future releases of Prusa Slicer and FW. This thread is a duplicate of #1659, so I'll close this specific thread as duplicate, but feel free to add more comments on the original thread in case! 

Alessandro Pantaleo
Prusa Research

closed 2022-12-01 by Prusa-Support