prusa3d/Prusa-Firmware-Buddy #2772: [BUG] With 4.4.0, Mini loses wifi connection over time

mix579, 2022-11-27:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - [e.g. 4.4.0 final]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint**
  Prusalink

**Describe the bug**
Didn't have any issues at all with wifi in 4.4 beta and RC1. Now a couple of days after installing RC1, the Mini has twice already lost its wifi connection. Mini is on pretty much all the time, and it seems that after a few hours the connection just drops (wifi symbol goes gray, no more IP address). Reboot fixes it, immediately reconnects.

luciofncnd, 2022-11-28:
The same happens to me, but after a couple of minutes of starting the printer. I installed the 4.4.0 release version and the wifi ESP01S module on my Mini+. It manages to get Wifi connectivity when turning on the printer, but it looses it after a couple of minutes, and never gets back. Only a full printer restart brings back the connectivity. It didn´t happen when using ethernet with previous firmware versions.

Eldenroot, 2022-11-28:
Same here even with final 4.4.0

ShadowDrakken, 2022-12-14:
Not sure if this is related, or a different issue: I have to restart my mini several times before Wi-fi connects _at_all_

Prusa-Support, 2023-01-12:
Thanks for reporting.

We are aware of connection issues caused by the ESP module not resetting correctly and causing other possibly related problems.
The firmware release 4.4.1-RC1 fixes that, improving the Wi-Fi connection, and therefore this issue could be closed as solved.

Now the ESP Wi-Fi module behavior should be even more reliable with FW [4.4.1-RC2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1-RC2).
Feel free to test it and share feedback.

Michele Moramarco
Prusa Research

czredhat, 2023-02-08:
It looks that 4.4.1-RC1 doesnt solve this issue. PrusaLink stops after a while even in this version.

Prusa-Support, 2023-02-16:
The RC2 was improved further but doesn't matter now as the final 4.4.1 is already available.
Please try the latest stable version.
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1

All in all, the connection may get lost from time to time, and the _Wi-Fi auto-reconnect_ feature was set in place to make it up for it.
When the connection is lost, the ESP Wi-Fi module will attempt a new connection every 60 seconds.
The connection reliability was already improved recently and could be further improved in the future.

In this case though, we may be looking at too generic problems, like the ones mentioned in issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2686.
We have multiple generic issues about connection troubles, probably too generic for us to focus and improve a specific aspect of the firmware.
I recommend closing this issue and keeping on listing generic connection faults in issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2686.

Michele Moramarco
Prusa Research

plandevida, 2023-05-20:
Hi,

I have experience all sort of issues with the WiFi on the Mini, here are my conclusions and I hope it will help to reproduce the issues.

Regarding the Wifi signal strength, I noticed having the ESP WIFI module inside the buddy case make the signal really weak and loses connectivity all the time. So, I put the chip outside the printer and now my signal is stable.

Nevertheless, I get so many issues with the Prusalink:
1) After printing: I send a file to print and after finishing I cannot send another because the prusalink does not respond. Reseting the printer it rarely fix the issue, just shutting down the printer works, and not always; I need to wait for a while to turn it on.

2) After cooling dow: sometimes I am able to send several print files, but if I wait until the printer start cooling down automatically, mostly every time the link is broken again. I cannot revive the service by turn off/on the service in the settings or turn off/on the WIFI interface.

Prusa-Support, 2023-06-15:
Recognizing some kind of patterns is probably the best alternative to reproducibility steps but, so far, we still can't controllably reproduce the issue.
The pattern here may also find an explanation in the Wi-Fi signal refresh rate: if the connection is lost a new connection is attempted once every minute.

Thanks for sharing your experience and we will keep investigating.

Michele Moramarco
Prusa Research

GithubUser99999999, 2024-04-28:
Happened with 5.1.0 as well and now with 6.0 the WiFi isn't even connecting anymore at all after the initial first connection after updating WiFi. It's kinda sad that it's basically impossible to use the Prusa Mini+ with WiFi since ever.

Seaborz, 2024-07-04:
For me the prusalink website also becomes unavailable. Updated to 6.0.3 but still the same issue. However, when i have a continuous ping open everything keeps working. I found this out by accident. Not a solution, but maybe some of you can verify this?

Prusa-Support, 2024-09-13:
This issue may actually be closed as obsolete because it is based on a very old and deeply different firmware however, before closing, I'd like to point out that [FW 6.2.0-alpha1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-alpha1) includes _Wi-Fi stability improvements_ among other things.

Please give it a try if you don't mind and consider closing the issue.
The final stable release should be available soon and rest assured that connectivity and data flow optimizations remain part of the regular firmware improvement routines. 

Michele Moramarco
Prusa Research

closed 2024-11-26 by github-actions[bot]