prusa3d/Prusa-Firmware-Buddy #2781: [BUG] Mini connects to network (via DHCP) but is unreachable

barrettj, 2022-11-30:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - Mini

**Printer firmware version** - Latest RC

**Original or Custom firmware** - Original

**Optional upgrades** - ESP01S

**Describe the bug**
Gets an IP via DHCP and says everything is good with the WiFi but no devices can connect

**How to reproduce**
Just try to use it

**Expected behavior**
I can connect from other computers on my network via IP address in the web browser or via Prusa Link.  I can see it in my Router's list of devices.

barrettj, 2022-11-30:
Similar to Couf's experience on the forum: https://forum.prusa3d.com/forum/hardware-firmware-and-software-help/prusa-mini-mini-wifi-troubleshooting/#post-615559

diminDDL, 2022-11-30:
I've been having similar issues with the RC. Mine appears to work for a minute or two after connecting to the network after which it becomes unresponsive to the web page, prusaslicer or pings.

Prusa-Support, 2022-12-26:
Thanks for reaching us.
We are aware of connection issues caused by the ESP module not resetting correctly and other possibly related problems. 
The firmware [4.4.1-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1-RC1) should improve the Wi-Fi connection.
Feel free to help the testing and share feedback

Michele Moramarco
Prusa Research

diminDDL, 2023-01-01:
@Prusa-Support Thank you for the heads up, flashed the RC today and the connection now seems rock solid and the interface is also a lot more responsive than it was with 4.4.0, good job :+1:

barrettj, 2023-01-05:
I also had time to install the 4.4.1 RC firmware this week and can also confirm it's working great

closed 2023-01-05 by barrettj