prusa3d/Prusa-Firmware-Buddy #2788: [BUG] M114 causes stuttering

pauhull, 2022-11-30:
Because there is no implementation of M154, hosts need to poll the current position with M114. Because the Mini firmware is based on an outdated version of Marlin, M114 waits for the motion planner to synchronize and thus causes stuttering.

A solution would be to update M114, implement M154 or extend M155 with the C parameter, like it is done in the i3 firmware.

Eldenroot, 2022-11-30:
I believe there will be Marlin update in the next release (hoping for 4.5.0)?

https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/2488

mkbel, 2022-12-17:
https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/2488 is definitely not making it into 4.5.0. I just managed today to start first print with Marlin updated to 2.1.x. Most of the features ale still broken with updated Marlin and 4.5.0 is already feature frozen going trough bug-fixing only. So Marlin will be updated in some later release.

closed 2024-06-19 by github-actions[bot]