prusa3d/Prusa-Firmware-Buddy #2789: [ENHANCEMENT] AutoExec / AutoLoad .gcode

rinkek, 2022-11-30:
**Prusa Mini**

Would be nice if we can have a .gcode file that's always loaded from USB drive at start, just like a AutoExec.bat in DOS.
This way we can run any settings everytime we start the Mini.

Why?
I'm running a directdrive extruder, which has a much shorter filament load/unload.
Also has my pinda probe an offset of 14mm on y-axis.

This is just an example, but i can think of many more people would like to run at start and gives people the freedom to change what they want evrytime the printer is started.
This way nothing in eeprom is changed and defaults back to stock when file is removed from USBdrive.

At this time i'm running a start.gcode everytime my printer is started (i guess many do), but sometimes i seem to forget to run this.

murk-sy, 2022-12-02:
You should be handling that within sliced gcode files as printer's start gcode. Autoexec usage isn't exactly common anymore and for a good reason.

As a workaround, you could manually modify a gcode file to have a future file modified date, making it always the first to be selected when plugging in the drive. Or, modify firmware to your needs and compile it yourself.

Prusa-Support, 2022-12-27:
Thank you for your suggestion.

Such a feature shouldn't be necessary for the normal printer using stock parts. Also, this is usually handled via Prusa Slicer (_printer settings > custom g-code > start g-code_).
An option like that could be handy though.

Our developers will make their considerations.

Michele Moramarco
Prusa Research

rinkek, 2022-12-29:
Thanks Michele!
The reason is that we cannot use the M500 command on the Mini. I need this as i am using a LGX extruder which has its pinda probe offcentered. This means when it's doing the leveling it will measure off the bed, resulting a nozzle digging into the bed with force. An autoexec could prevent this (or bring M500 back)

I know my Mini is heavily modified (Mini Bear) but that's the fun with this little printer.

closed 2024-06-19 by github-actions[bot]