prusa3d/Prusa-Firmware-Buddy #2794: [BUG] Wrong remaining time

ThomasCZ, 2022-12-03:
**Printer type** - MINI

**Printer firmware version** - 4.4.0

**Original or Custom firmware** - Original

**Describe the bug**
Remaining time in PrusaLink is wrong. I do not think, this is bug in PrusaLink, the printer is probably providing wrong time.

[image]

Prusa-Support, 2022-12-27:
Thanks for reporting.

Does that happen with a specific print only?
Does the problem persist with [FW 4.4.1-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1-RC1)?

Michele Moramarco
Prusa Reseearch

nighthawk8301, 2023-03-07:
+1 

I got this bug in the 4.4.1 release version, while printing some of the Prusa mini printable parts.
After a few minutes the bug was healing itself and the correct remaining time was displayed again.

Prusa-Support, 2023-03-28:
As described at https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1321#issuecomment-1452345163, there are various reasons for the clock and the print time estimate not being extremely accurate, or temporarily offset.

The auto-heal is actually a sign that code `M73` regularly occurs during the print (expected behavior).

Because the described problem is too generic and seems actually related to the printer's firmware, this issue should be closed as a duplicate of issues https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1067 or https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1321.

Michele Moramarco
Prusa Research

closed 2023-03-28 by Prusa-Support