prusa3d/Prusa-Firmware-Buddy #2795: [BUG] Printer shows wrong print end time

TheSithCat, 2022-12-03:
**Printer type** - [MINI+]

**Printer firmware version** - 4.4.0

**Original or Custom firmware** - Original 

**Describe the bug**
  Printer show wrong print end time, around hour less. Prusa Link shows correct end time

[image]
[image]

TheSithCat, 2022-12-04:
[image]

TheSithCat, 2022-12-04:
#136

vanboom, 2022-12-04:
SUGGESTION:  remove the print ending time and restore the "Time Remaining" metric.  99% of the time I look to the screen to see how much time is remaining and that metric is now gone.

closed 2022-12-07 by TheSithCat