prusa3d/Prusa-Firmware-Buddy #2798: [BUG] <4.4.0 release issues with mesh bed leveling

prongato, 2022-12-05:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - [e.g. 4.0.5, ...]

**Original or Custom firmware** - [Original/Custom]

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint

**Describe the bug**
  A clear and concise description of what the bug is.

**How to reproduce**
  Please describe steps to reproduce the behavior.

**Expected behavior**
  A clear and concise description of what you expected to happen.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

Prusa-Support, 2023-01-07:
Thanks for reporting this.

That's very interesting. This issue could be a duplicate of issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2729, or related to it.
The missing output may also be related to a known issue with serial communication (https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2815). Please try [FW 4.4.1-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1-RC1).

Moreover, it is curious how in issue #2767, both serial communication and calibration issues are mentioned too.
I may start thinking that the two things are related.

Regardless of the missing probed values output, would the Mesh Bed Levelling be successful?
Would the first layer accuracy be affected in any way?

Michele Moramarco
Prusa Research

ZasoZ, 2023-01-11:
I had the same problem with 4.4.0 but downgraded to solve the issue , if you need an additional persons i out i can upgrade to 4.4.0 again and see if the first layer is affected.

Kind regards

Prusa-Support, 2023-01-21:
Yes, more tests could help. We may need more printers and users to be able to reproduce the issue.
Also, I wonder if this is only a missing value output or if the calibration and the first layer quality are affected in any way.

Please try FW [4.4.1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1) (final) though, as it fixes some serial connection problems.

Michele Moramarco
Prusa Research

ZasoZ, 2023-02-02:
Hi So now that i have had time to test it and i can report that mesh bed leveling fails in the calibration menu as well. Though updating to 4.4.1 fixed that issue and my problems with G29 (and octoprints mesh bed visualizer). So happy to report that it is fixed for me.

Prusa-Support, 2023-02-14:
Thanks for your feedback. That confirms the nature of the problem we are looking at, which is fixed by now.

I haven't heard from other users since the possible issue was posted more than one month ago but it seems safe to close this issue as solved. Feel share more feedback though.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-06-03:
Sorry for re-intervening in this closed issue.
According to a new issue (https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3036), this problem continues happening on the most recent firmware.

I wonder if any of the users in this conversation is still able to reproduce the problem with the most recent firmware version, or if any user recognized a sequence of troubleshooting steps that probably resolved the problem in their case (e.g. a system refresh, disconnection of Octorpint, the use of a different USB cable...).

Michele Moramarco
Prusa Research

Tommy544, 2023-06-09:
I am still experiencing this problem using firmware 4.4.1 on my Mini+.

In my case, I get missing entries more often than the correct output and my workaround so far has just been to "brute-force" it and start the measuring process enough times until I get a complete output.

I was using the latest version of Octoprint on a Raspberry Pi 3. I haven't tried to switch USB cables to see if that has any effect.

rinkek, 2023-06-21:
Yes, i still have this issue in 4.4.1 on the Mini+
I've only tried this with Octoprint. You can see it in the terminal; missing values.

marek26340, 2023-06-22:
Hi. I have already made some comments in other issues here, detailing my issues with the firmware version 4.4.1. #3042 (2nd comment)

So yes, I also have discovered this issue, among with other things, all of which disappeared the moment I downgraded back to 4.3.4.
Feel free to delete my comment if you think that I am spamming. I'm only confirming that I did encounter this issue.

n-dony, 2023-07-02:
Hi. Same problem. Randomly missing values ​​in the bilinear leveling grid. Mini+ factory assembled, firmware 4.4.1. 
16 points taken without error on the printer, but sometimes missing values in ​​the output to the terminal. 
Could it be a wrong mesh in the printer memory ? Is it possible to dump the saved mesh from the printer ?

Prusa-Support, 2023-09-23:
Feel free to give it a go with the new FW 5.1.0 but please mind it is an alpha version.
A factory reset may be needed.

However, our developers plan to focus on serial line communication issues in the next few firmware releases (probably after 5.1.0) so please let us know if the problem persists.

Michele Moramarco
Prusa Research

jakl-159, 2023-11-22:
Printer type - MINI/MINI+

Printer firmware version - 5.1.0 BETA

Original or Custom firmware - Original

Optional upgrades - Prusa Filament Runout Sensor, Octoprint

USB drive or USB/Octoprint - micro USB/Octoprint

I do have the same problem with my Prusa MINI+. After doing maintenance I wanted to to check the bed level but getting similar response as above/other threads on this issue (MBL). This is the response after G28 -> G29 commands from pronterface/octoprint:

`Default storage slot 0 selected.
Mesh invalidated. Probing mesh.
echo:endstops hit:  Z:-0.10
echo:endstops hit:  Z:-0.11
echo:endstops hit:  Z:-0.11
echo:endstops hit:  Z:-0.13
echo:endstops hit:  Z:-0.15
echo:endstops hit:  Z:-0.14
echo:endstops hit:  Z:-0.10
echo:endstops hit:  Z:-0.10
echo:endstops hit:  Z:-0.10
echo:endstops hit:  Z:0.03
echo:endstops hit:  Z:0.03
echo:endstops hit:  Z:0.03
echo:endstops hit:  Z:0.13
echo:endstops hit:  Z:0.13
echo:endstops hit:  Z:0.13
echo:endstops hit:  Z:-0.12
echo:endstops hit:  Z:-0.12
echo:endstops hit:  Z:-0.12
echo:endstops hit:  Z:-0.23
echo:endstops hit:  Z:-0.23
echo:endstops hit:  Z:-0.23
echo:endstops hit:  Z:-0.35
echo:endstops hit:  Z:-0.34
echo:endstops hit:  Z:-0.34
echo:endstops hit:  Z:0.01
echo:endstops hit:  Z:0.01
echo:endstops hit:  Z:0.01
echo:endstops hit:  Z:-0.02
echo:endstops hit:  Z:-0.02
echo:endstops hit:  Z:-0.02
echo:endstops hit:  Z:-0.10
echo:endstops hit:  Z:-0.11
echo:endstops hit:  Z:-0.11
echo:endstops hit:  Z:0.13
echo:endstops hit:  Z:-0.09
echo:endstops hit:  Z:-0.09
echo:endstops hit:  Z:-0.09
echo:endstops hit:  Z:-0.17
echo:endstops hit:  Z:-0.17
echo:endstops hit:  Z:-0.19
echo:endstops hit:  Z:-0.19
echo:endstops hit:  Z:-0.19
echo:endstops hit:  Z:-0.24
echo:endstops hit:  Z:-0.24
echo:endstops hit:  Z:-0.24
Extrapolating mesh...done
Unified Bed Leveling System v1.01 active`

ricarva, 2023-11-30:
Hi, everyone

I'm experiencing this issue, apparently: it manifests by Octoprint's Bed Level Visualizer not being able to get/refresh values, even though the procedure completes on the printer with no errors.

Is this working for anyone? Is it universally broken?

marek26340, 2023-11-30:
> Hi, everyone
> 
> I'm experiencing this issue, apparently: it manifests by Octoprint's Bed Level Visualizer not being able to get/refresh values, even though the procedure completes on the printer with no errors.
> 
> Is this working for anyone? Is it universally broken?

Yes, it's a widely known problem.

Go ahead and upgrade to 5.1.0, it's supposed to finally fix it. 4.4.0 was just plain unusable in my opinion. It's been great so far!

marek26340, 2023-11-30:
> Still ocuring in 5.1.0 release for me.

Oh. Goddamnit! 
Should I just delete my comment then?

ricarva, 2023-11-30:
> Feel free to give it a go with the new FW 5.1.0 but please mind it is an alpha version. A factory reset may be needed.
>
> However, our developers plan to focus on serial line communication issues in [...]

@Prusa-Support 

Yup, still present in 5.1.0

n-dony, 2023-12-02:
Hi,
For me, no more missing values ​​with FW 5.1.0. The initial issue is close for me.
But I need to ask for a report with G29 T to get a matrix usable by the plugin. 

G28 ; home all without mesh bed level
G29 ; mesh bed leveling 
@BEDLEVELVISUALIZER
G29 T; mesh bed leveling report

closed 2024-06-19 by github-actions[bot]