prusa3d/Prusa-Firmware-Buddy #2815: [BUG] after update to 4.4.0

svdstoep, 2022-12-07:
**Printer type** - Prusa Mini+

**Printer firmware version** - 4.4.0

**Original or Custom firmware** - Original

**USB drive or USB/Octoprint** - USB Octoprint to Windows host

**Describe the bug**
After the upgrade to 4.4.0 the USB connection is not recognized in Windows (tested with Server 2019 and Windows 11)

**How to reproduce**
Connect to USB cable to a Windows host and power on printer. In device manager you can see a new unkown device; **USB Composite Device**, where the old 4.3.4 firmware says USB Serial Device. In 4.4.0 BETA2 the USB connection works.

[image]

afonsosantos, 2022-12-10:
Same issue here, Prusa Support says Prusa Mini is not Windows compatible, which is false.
Waiting for some triage on this issue.

thomasvnl, 2022-12-11:
Same issue here on latest version of Windows 10.

TheMoen, 2022-12-11:
Same issue here. downgraded to 4.3.4 to do what i was plannling on testing out. That worked fine. when upgrading back to 4.4.0 the "USB Composite device" "could not start".

Robotdevul, 2022-12-13:
I have the same issue with firmware 4.4.0

MistrKernnunos, 2022-12-20:
Hi, this PR should fix it. #2844

svdstoep, 2023-01-11:
I can confirm the USB-Connection works again  with 4.4.1-RC2. Will do some more testing in the following days.
Thank you!

Prusa-Support, 2023-01-27:
Thank you for your feedback.
The [final FW 4.4.1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1) has been released last week so the issue should definitely be fixed now.

I'll close this issue as solved but please feel free to provide further feedback.

Michele Moramarco
Prusa Research

closed 2023-01-27 by Prusa-Support