prusa3d/Prusa-Firmware-Buddy #2816: Prusa Mini Firmware 4.4.0 causes preheat error with E3D's Revo Micro

FrHePo, 2022-12-08:


Robotdevul, 2022-12-13:
I've had the same issue with my Revo while heating up after updating the firmware to 
4.4.0  I reverted back to 4.3.4 and have not had the issue.  The revo has been installed for several months with no such errors until the new firmware.

Moyrn, 2022-12-18:
Hello. Same problem here. I faced the issue since 4.4 and had the revo 6 micro installed (rev 1) with the micro fan. 
i finally upgraded to rev2, with stock fan.

I still have the problem...

Thankee

FrHePo, 2022-12-18:
That's hard to understand. With the stock Prusa-fan you shouldn't have this problem. Are all 3 pins connected to the board? Maybe there's a broken wire somewhere? I would check the fan-wiring from the fan to the board. If possible use a multimeter.

Moyrn, 2022-12-18:
Hello

Yes, very strange. I admit.
Yes the 3 pins cable is plugged at its normal place. I could do some
printing with this new configuration. 2 or 3.
Seems like problem arises when preheating for loading/unloading cable. Most
probably just random.

It seems the read T°C value varies a lot during the (pre)heating. Maybe
something got wrong

Maybe it's a deffective revo heater because of the too poor micro-fan i had
for 7 months. Pretty weird.

FrHePo, 2022-12-18:
Just after the 1st Revo's were shipped there were reports of defective Revo heaters. I got one too that was defective and it got replaced on guarantee. I don't think it has anything to do with the original fan. I have 2 Revo's running with original 2-wire fans without problems. I like those 10mm fans because they keep the extruder light and this is especialy positive for the Mini.

FrHePo, 2022-12-18:
BTW, when I remember correctly it's not the heating itself that was defective but the temperature sensor. Since the temperature sensor is built in in the heating coil, it's the complete heating unit that needs to be replaced. The symptoms you describe seem to point to a defective temperature sensor.

Moyrn, 2022-12-19:
Indeed. Thanks. I'll contact my vendor (3djake)

Prusa-Support, 2022-12-20:
Thanks for reporting.

You may already know that the _preheat error_ would be triggered if the target temperature isn't reached, precisely if the temperature isn't risen by at least 2ºC over 20 seconds.

Is the _other layers_ temperature significantly higher than the _first layer_ temperature?

Is the print fan disabled for the first layer?
How many layers does it take for the fan to reach the maximum speed, and what is the maximum speed?

Does the printer pass the self-test, especially the nozzle heating test?
Can the printer reach ASA (260°C) or a higher temperature?

It may help to get the print file and a video of the error (starting about 20 seconds before the occurrence).

Michele Moramarco
Prusa Research

FrHePo, 2022-12-21:
See: https://forum.prusa3d.com/forum/hardware-firmware-and-software-help/firmware-4-4-0-causes-preheat-error-with-e3ds-revo-micro/#post-628889

Moyrn, 2022-12-21:
Hi all.

On my case, standard print. Yes selftest OK 3 days ago. Only the print fan
test was ko with initial e3d micro fan. Moved to rev2, meaning back to
stock fan + revo and self test OK.

I'm presently abroad for a while so no video. Coding on others for this.
Never printed in Asa or abs on the mini.
Maybe petg 235 max (sunlu)

Aside this preheat error it seems I can still print as I did one on at 205
last week.
All I can say is that it seems that read temp is fluctuating too much. As
if with a loose thermistor... Which is nonsense with the revo.

So as said above, maybe a defective unit... Revealed by 4.4.0 preheat test?

I'll contact e3d for warranty

Thanks for taking the time

Prusa-Support, 2022-12-27:
@FrHePo. So far it may all seems to be an issue with the specifically used fan. I'm glad that you figured it out and thanks for the feedback.

@Moyrn. Fluctuating temperatures may surely cause similar issues. This may be a different issue but keep us posted after the replacement if you don't mind.

Michele Moramarco
Prusa Research

Moyrn, 2022-12-27:
Hello. Aye. I will receive the new unit this week. Today i got "preheat
errors" several times, then i finally could print. Made a video where we
see crazy fluctuations 230>150>210>240>170...
I tried to pour some glue on the thermistor "hole" (after dismanting the
revo heater). Not really better.
Sent an email to 3DJake > told me to contact E3D > contacted E3D : no
answer yet

I'll let you know. thanks

Moyrn, 2022-12-29:
Hello all.

Did not perform the switch yet.
Just to let you know that after watching videos (showing erratic temp
reading wile preheating), E3D recognizes the heater as defective. I'm now
turning to 3DJake... (again).

Take care

Wilfried

Moyrn, 2023-01-02:
Hello

Happy new year to you all ! I hope I'll get my XL this year ...

Back to our topic. I just installed a brand new REVO Heater I purchased
(i'm still waiting for the replacement part from 3DJake).
I made a preheat, filament load, filament unload.
All went fast and fine. I checked and saw no erratic behaviour of read
temperatures...

I hope this helps

Wilfried

closed 2024-06-19 by github-actions[bot]