prusa3d/Prusa-Firmware-Buddy #2820: Measured temperature is not matching expected value

dsmccusk, 2022-12-10:
Mini+ 4.4.0 w Filament Sensor

Brand new printer getting the "Measured temperature is not matching expected value." trying to perform the initial test print.  Always within about 5 minutes of reaching target temp (PLA - 215C).  I did the first time calibration with the previous firmware and then upgraded to 4.4.0 and started experiencing this after the upgrade.  I've checked and reseated connections.  Loaded/unloaded filament, cleaned the hotend.  Everything short of replacing parts.

svdstoep, 2022-12-10:
I think it's a new notice in the 4.4.0 firmware. I've got this issue multiple times (also with brand new thermistors), the problems were always in the thermistor. Broken cable (invisible to the eye) or just not within the specifications.

I would suggest to first try to measure the resistance of the hotend thermistor
https://help.prusa3d.com/article/multimeter-usage_2117?gclid=Cj0KCQiA1sucBhDgARIsAFoytUtSRxlKZ9NW8_BUnLwevIXIrO2_5E8mZkNVbFmMPJpcdrJKivNRnWgaAmT1EALw_wcB

mix579, 2022-12-10:
I've reported it under [#2768](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2768). It's super annoying. Never happened before I switched to 4.4.0. My workaround is to first heat the nozzle up to 200, which seems to be fine, and then from there start the filament load or raise temperature manually. But this feels like either a bug or a super aggressive algorithm. Would be nice to at least be able to turn the warning off.

Prusa-Support, 2022-12-10:
Thanks for reaching us.

Please allow me to close this issue as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2768.
Feel free to continue the conversation on the other issue or let me know if you see reasons for the issue to stay open separately.

Michele Moramarco
Prusa Research

closed 2022-12-10 by Prusa-Support