prusa3d/Prusa-Firmware-Buddy #2821: [Plausible BUG] Stuttering when moving fast

TheMoen, 2022-12-11:
**Printer type**  
Prusa Mini+

**Printer firmware version** 
4.4.0

**Original or Custom firmware** 
Original

**Optional upgrades** 
 Filament Runout Sensor
Revo micro hotend

**USB drive (not relevant)**
Original USB drive

**Describe the bug**
When the printer is performing fast travel moves, it is stuttering.
It did happen at the very end of first layer calibration both Y Axis and X Axsis
I also did happen when printing.

**How to reproduce**
This happend when upgrading from firmware 4.4.0 RC1 to firmware 4.4.0

**Expected behavior**
I expected the movement to be smooth

**G-code**
At the end of first layer calibration

**Solution for me**
Downgrading to firmware 4.3.4 then upgrading to 4.4.0
(The reason i did it was to use pronterface, since the USB driver does not work with 4.4.0)
This solved the issue and the printer is running fine on fast travel moves.

Eldenroot, 2022-12-11:
I am glad that I am not the only one. I thought it is a mechanical issue. I had to downgrade to 4.3.4 back...

Eldenroot, 2022-12-11:
The worst thing it happens during printing as well

TheMoen, 2022-12-11:
> I am glad that I am not the only one. I thought it is a mechanical issue. I had to downgrade to 4.3.4 back...

Did you also upgrade from 4.4.0 RC1 to 4.4.0?

Eldenroot, 2022-12-11:
Yes and no issue with RC1

Eldenroot, 2022-12-12:
This is quite critical :(

quadeare, 2022-12-18:
I confirm this behavior. I've got the same issue with my Prusa mini with FW 4.4.0.

Prusa-Support, 2023-01-14:
Thanks for reporting.

So upgrading from 4.4.0-RC1 to 4.4.0 causes the issue.
Upgrading from 4.3.4 to firmware 4.4.0 solved the issue, right?
Would a factory be an equally effective fix?

This may actually be related to https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2326 or https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2333 where the printer movements were altered after the upgrade.

However, FW [4.4.1-RC2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1-RC2) is out.
Please give it a try and provide feedback if you don't mind.

Michele Moramarco
Prusa Research

TheMoen, 2023-01-14:
"So upgrading from 4.4.0-RC1 to 4.4.0 causes the issue."
Yes
"Upgrading from 4.3.4 to firmware 4.4.0 solved the issue, right?"
Yes
"Would a factory be an equally effective fix?"
Guessing factory reset.. I belive i did that, and it did not solve the issue,

However 2 weeks after upgrading from 4.3.4 to 4.4.0 this issue came back. 
That made me downgrade to 4.3.4 where there were no issues.

I am currently running 4.4.1-RC2, have not used it very much, I have had no issues with the two prints i have done.

Prusa-Support, 2023-01-24:
Sounds good. I'd suggest using [FW 4.4.1 (final)](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1) as it was freshly released last week.
I'd like to see positive feedback on the final release before closing the issue, thanks.

Michele Moramarco
Prusa Research

TheMoen, 2023-01-26:
I am soon done with my first print with FW4.4.1(final), It seems fine, have not heard any stuttering.

TheMoen, 2023-02-18:
This is still an issue on v4.4.1 it happened today, solved by restarting the printer.

TheMoen, 2023-02-18:
It seems to happen after doing a few first layer calibration.
Video:
[https://youtu.be/EN_CkSNqmUs](url)

Prusa-Support, 2023-03-03:
Thanks for the feedback.

It seems to be an intermittent issue that I'm not able to reproduce myself.
Our developers will look into this.
Meantime let's please keep on monitoring the situation and hearing from other users.

Michele Moramarco
Prusa Research

Ellipsre, 2023-05-02:
I am having the same issue running 4.4.1 all stock PRUSA Mini+. Issue triggers while performing 1st layer cal.

Prusa-Support, 2023-05-31:
Would a factory reset solve the problem?
If not, would temporary downgrading to 4.3.4, and re-upgrading to 4.4.1 after (or 4.4.0 as some users did) remove the problem?

Michele Moramarco
Prusa Research

kozross, 2023-06-01:
I've witnessed this intermittently, mostly on Y moves. Also on 4.4.1, haven't tried downgrading. Seems to go away after some time though.

BrentTechDesigns, 2023-08-01:
I've been running 4.4.1 for a couple of months now with no issues.  However, I just had to replace the Revo heater core due to a bad thermistor.  Upon successfully doing so, I now notice this stuttering during fast travel.  

I googled and found the issue posted here.  I performed a power cycle and it appears to be back to normal with no stuttering during fast travel.

kozross, 2023-08-01:
I can confirm stuttering happens when I do any of the following:

* Run first layer calibration upon the print completing, when the toolhead parks.
* When I use [this linear advance test](https://ellis3dp.com/Pressure_Linear_Advance_Tool) (with Marlin settings).
* When I use [this linear advance test](https://marlinfw.org/tools/lin_advance/k-factor.html). This one ends up shaking my printer so violently it actually ruins the test.

RandomDucklingGenerator, 2023-11-18:
Stock Mini+ with 4.4.1. Happens when calibrating first layer.

https://youtube.com/shorts/9CUJFUPbRZk

notacoat, 2024-01-11:
I rebuilt my printer and started having this error on both x and y axis when traveling fast. It does not happen when homing or running bed mesh.
It oddly resolved itself yesterday, which makes me think my printer isn’t square or something. The hotend was in the top left corner and i homed, causing the hotend to move diagonally to the bottom right (x and y moved at the same time.) after that x and y axis are moving without the stutter noises

closed 2024-06-19 by github-actions[bot]