prusa3d/Prusa-Firmware-Buddy #2822: [BUG] filament-runout-sensor not recognized when using Prusalink

ErGeeBee, 2022-12-11:
**Printer type** - [MINI]

**Printer firmware version** - [e.g. 4.0.5, ...]

**Original or Custom firmware** - [Original/Custom]

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint

**Describe the bug**
  A clear and concise description of what the bug is.

**How to reproduce**
  Please describe steps to reproduce the behavior.

**Expected behavior**
  A clear and concise description of what you expected to happen.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

Prusa-Support, 2023-01-14:
Same as https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2334, I'm afraid that Prusa Link can't reproduce the printer's typical error messages and warnings at the current stage.

Thanks for reporting. Our developers will work on this.

Michele Moramarco
Prusa Research

HaWiWe, 2023-03-29:
It could also be that the sensor is disabled in the settings, that was the problem with mine. It resets the settings every reboot though. 😑

Prusa-Support, 2023-06-30:
Sorry for the late reaction.
Recently the presence of a bug making the filament sensor turn off (under specific circumstances) seems to be confirmed.
Please read more at https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1833#issuecomment-1585040893. 

As previously mentioned, the lack of printer statutes/events notifications through Prusa Link instead is a known limitation at the moment.

Please allow me to close this issue since it partially duplicates other more specific and progressed ones.

Michele Moramarco
Prusa Research

closed 2024-06-19 by github-actions[bot]