prusa3d/Prusa-Firmware-Buddy #2828: [BUG] Filament Auto Load Not cooling once loaded

seifer69er, 2022-12-14:
On the latest 4.4.1 RC1,

Enabling the filament autoload, it was noted that once successfully loaded, the nozzle temp and heatbed stay in the heated state.
These should return to cooldown

Tested by unloading PLA filamnet and loading PETG directly after.

Cadsters, 2022-12-16:
Same issue here, perhaps good  to mention this in the releasenotes.

Eldenroot, 2022-12-16:
Prusa team should add some info and fix this

Prusa-Support, 2023-01-09:
Hello! Thank you for the report, this is expected behavior, but I reported it to devs for them to evaluate a possible change on this for a future release, for example a possible idea could be having a choice to either keep it heated or not - though this is not yet implemented, nothing we can promise, but it's reported and known from our side. 🙂 

This thread is almost a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1331 (though that one referred to unload, fair point), also it's more loosely related with https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/706. 

In case of more info from our side, we'll surely update on these two threads. 

Alessandro Pantaleo
Prusa Research

closed 2024-06-19 by github-actions[bot]