prusa3d/Prusa-Firmware-Buddy #2834: [BUG] Print time finish is not recalculated after M600 filament change

Eldenroot, 2022-12-16:
**Printer type** - [MINI+]

**Printer firmware version** - [e.g. 4.4.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor + Wifi module + Bondtech IFS extruder]

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  Time (finihs print time) should be recalculated when you change the filament color - now it does NOT take "paused" time for changing the filament, so the time is not correct. 

**How to reproduce**
  Use multicolor print, check the timer.

**Expected behavior**
  Finished time should be recalculated - it should plus the time when the printer was paused because of changing the filament.

**G-code**
  ---

**Crash dump file**
  ---

**Video**
  ---

thijstriemstra, 2022-12-17:
>  now it does NOT take "paused" time for changing the filament, so the time is not correct.

Interesting bug 😄 So theoretically, if you leave the printer in change filament state for an hour or two, and the total print time is an hour, it would show a negative remaining time after changing filament?

Eldenroot, 2022-12-17:
Yes, at least now it was like this...

I use ethernet connection (wifi is unstable).

thijstriemstra, 2022-12-17:
there's a 4.4.1 rc1 available, maybe that'll fix it.

Prusa-Support, 2023-01-17:
Thanks for reporting.

Theoretically, a pause won't permanently alter the remaining time estimation, because it is updated every time code M73 occurs during the print.
I can speculate that the remaining time auto-fixed sometime later during the print, unless the gcode didn't include M73 codes for some reason.

The erratic end-time calculation (based on remaining time, available upon network connection) should be fixed with the adoption of the Real Time Clock (RTC), which has been tested and further polished since the first implementation.
Feel free to test [FW 4.4.1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1) and share your feedback.
We have other dedicated issues about incorrect time calculation though.

Please make sure that M73 codes are included after M600 so that the remaining time estimation could be restored sometime during the print.
Can the issue be closed?

Michele Moramarco
Prusa Research

Eldenroot, 2023-01-17:
M73 should be added automatically not? I use Prusa slicer and if not, it should be added when you use M600 command.

closed 2023-01-17 by Eldenroot