prusa3d/Prusa-Firmware-Buddy #2835: 4.4.0 Purge Filament functionality lost

GNNPNUT, 2022-12-16:
Comes up on the menu, but in faded text.  When selecting, nothing happens.

Asteec, 2023-01-17:
Exactly this is for me major problem, i have used this very frequently and now i cant.

Asteec, 2023-01-17:
And this issue unfortunately persist in new 4.4.1 firmware too.

Asteec, 2023-01-17:
> However, I have maybe different issue—when you click on “Purge filament”
> and click to “STOP”—the bed and the nozzle is still heating up. I think it
> should not…
>
> @Prusa-Support <https://github.com/Prusa-Support>
>
> Possibly there are two different bugs?

I think so, there are two bugs.

Asteec, 2023-01-17:
I have tested it, with filament sensor connected to the board while in
settings turned filament sensor to OFF, i have not the possibility to purge
the filament. But when i turn it in settings to ON, i have this possibility
and can purge filament. Before firmware 4.4.0 it was possible even without
filament sensor connected to the board. So it is a issue and may be
repaired. I put the sensor out before, because i have printed a lot of
flexible filaments and this sensor caused more friction and there were
problems with retractions. So the possibility to purge filament without
this sensor is necessary for a lot of people like me.

Asteec, 2023-01-17:
The problem you described when you click on “Purge filament” and click to
“STOP”—the bed and the nozzle is still heating up.
I tested it now. When i click on purge filament, it start heating, after 10
seconds, when i click STOP button, printer restarts itself and is not
heating up anymore. So yes, its absolutely a bug.

Asteec, 2023-01-17:
You can possibly reflash the firmwate again.

Eldenroot, 2023-01-17:
I have tried... still same.

So there are two bugs, quite annoying...

Prusa-Support, 2023-01-18:
Thank you for reporting.

As of now, the purge option is actually only available if the filament is detected (= filament sensor enabled and triggered). Not sure if this was an intentional change. Bug or not, this issue can stay open as an "enhancement request".

About the "stop" option, not setting the printer to cool down, I'm not sure if that behavior was changed. I believe that features like _load_ or _purge_ would trigger a preheat at the beginning, but would not set back the temperatures to "zero" if stopped.
I think this deserves a separate issue with an explanatory title, so that other users can find it more easily and our developers could have a neater view of the two issues.

Michele Moramarco
Prusa Research

Eldenroot, 2023-01-18:
@Prusa-Support - purge filament - it is a regression. Filament sensor is optional upgrade, so many users just are not able to use this feature? So how to purge nozzle for these printers? It should be changed.

About zhe second issue - there should be changed - one way - add another confirmation window after you click stop or the purge is completed... Or just turn off heating.

Asteec, 2023-01-19:
Not only filament sensor is optional upgrade, but even if you have one like
me, and its connected to the mainboard, if i can print some flexible
filament, i always turned filament sensor off in settings to lower friction
in the bowden style extruder system, to be able print flexible filament
without issue, but at the same time i will lose the possibility to purge
filament. Its only after update from 4.3.3 to 4.4.0 and persist to 4.4.1
firmware. So it can be alowed for every user to purge filament for whatever
reason it is. So please, return this option back.

GNNPNUT, 2023-01-19:
>     Do you use filament sensor?

Reflecting on this, I believe that I was able to fix the problem by RE-ENABLING the filament sensor.  When the upgrade was done, it disabled the sensor.  No issues with this since I made that change.  Surprised that it would have disabled the sensor, but I know the sensor was operational before making the upgrade as I had ran out of filament on a job a couple of days before.

Prusa-Support, 2023-01-26:
Yeah, that fixes it but our developers will consider enabling the "purge" option even when the filament sensor is disabled or not available.
This issue will stay open.

For the time being, please consider loading twice if the first time load was unsuccessful, or slowly extrude further via _settings > move axis_, or remove the upper brass nut to gain full manual control over the purge.

Michele Moramarco
Prusa Research

RAMilewski, 2023-04-25:
Purge Filament was also useful when changing filaments.  The button layout in the purge dialog inside filament change is poor.  After the question "Is the color correct?", you must rotate the knob 1 (and only 1) click to the right to get to "no" to trigger another purge cycle.  If you fumble and rotate 2 clicks the printer unloads the filament.  

Initiating a second purge quickly is sometimes important to prevent the extruded filament from cooling and having the second purge accumulate on the nozzle tip.

With the pre 4.4.0 purge code it was possible to answer "Yes" in the filament change dialog and then move to "Purge Filament" where there is no danger of accidentally unloading the filament.  This could be done about as quickly as moving to NO on change filament purge dialog.

The correct solution, is to fix the Purge Filament regression AND change the order of the buttons in the change filament dialog to:   " NO  YES  RETRY"

ChrisPirillo, 2023-09-14:
Brand new Prusa Mini+. Purge Filament greyed out like all others have reported. Right out of the box. ;)

MoneyJayMoney, 2023-10-10:
With 5.1 it completely ignores the filament sensor even though it is on and working when tested on the screen.  However, I've gotten to the end of two spools and it just kept printing.

closed 2024-06-19 by github-actions[bot]