prusa3d/Prusa-Firmware-Buddy #2836: [BUG] Perform a small extrusion prior to unload

Eldenroot, 2022-12-16:
**Printer type** - [MINI]

**Printer firmware version** - [e4.4.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  I use M600 a lot, unfortunately filament (new one, Prusament - so completely dry) stuck inside the extruder during unloading. The small tip blocks the output. Same issue was discussed for MK3 Prusa - 
https://github.com/prusa3d/Prusa-Firmware/pull/2318
https://github.com/prusa3d/Prusa-Firmware/issues/417

**How to reproduce**
  Unload filament (PLA)

**Expected behavior**
  Slower unloading

**G-code**
  ---

**Crash dump file**
  ---

**Video**
  ---

Prusa-Support, 2023-01-18:
Thanks for your suggestion.

This behavior was already improved in the past with FW [4.1.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.1.0) and [4.3.3](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.3).

This behavior may change though if the M600 command is not directly prompted by the firmware itself (for example if you are printing via serial line). I may seem to understand that you are regularly printing via USB flash drive though, so this shouldn't be a concern. Is that right?

Also, consider double-checking or replacing the hotend PTFE tube. If such tube is deformed, out-of-specs, or simply shaking together with the filament movements, it may cause irregular tips at the end of the filament.
Same applies to the hotend metal parts: even a tiny gap between heat-break and nozzle can cause irregular tips.

Michele Moramarco
Prusa Research

capi92, 2023-08-23:
Hi,

got the same issues on mk4 especially with 0.25 Nextruder nozzle. 
As is, 0.25 nozzle are not usable with m600. 
will check if immediate filament switch would improve, I used to be away thus coming back when printer is beeping I.e 30 / 60 seconds later, filament being cold can’t be pushed again in nozzle…

I have a small drop on top of the Nextruder nozzle that need nozzle to be dismantle to be cleaned up.

printing via usb.

closed 2024-06-19 by github-actions[bot]