prusa3d/Prusa-Firmware-Buddy #2837: Prusa Mini+ Blue Screen BSOD "media_prefetch. precise data access error"

GithubUser99999999, 2022-12-17:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI+]

**Printer firmware version** - 4.4.0 (worked fine since yesterday)

**Original or Custom firmware** - Original

**Optional upgrades** - Prusa Filament Sensor, ESP-01 WiFi module, Prusa 0.6 mm nozzle

**USB drive or USB/Octoprint**
  via LAN and USB drive (I always used the original Prusa Mini+ USB drive but with a USB extension cable; I also tried a different USB drive with the same result)

A PrusaLink tab is always open in my browser (since closing it would require reentering that password every time)

**Describe the bug**
  Every print stops suddenly and randomly while nothing else happens if I don't do anything. 

a) [shown on video below] When I hit "pause", the printer freezes, then restarts with a bluescreen of death ("media_prefetch. precise data access error"). Print failed.

b) [not shown on video below] When I instead remove the USB drive, the extruder moves into a safe position (only in 4 out of 5 cases) and the display tells me that the USB drive was removed or is interrupted (obviously) and then I reinsert it, hit resume and the print continues. The print is marred due to the interruptions. On one print it happened 7 times before I gave up. (Bear in mind that I tried this with or without an extension cable and with two USB drives. )

Is my USB Type A port broken? It's a bit wobbly from the start.

**How to reproduce**
  I guess I just did that already.

**Expected behavior**
  The printer working fine as usual.

**G-code**
  It happens with *every* print.

**Crash dump file**
  I have no idea where to find that.

**Video**

https://user-images.githubusercontent.com/101017273/208221308-3dc95210-17fe-426c-a2ec-291e613d5bd6.mp4

[image]
[image]

Prusa-Support, 2023-01-03:
Thank you for reporting the problem.

This issue may be related to issue https://github.com/prusa3d/Prusa-Link-Web/issues/316, or issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2640, or may even be a duplicate.

The interrupted print could actually be related to a faulty USB flash drive or USB port.
Please use a different USB flash drive, formatted FAT32.
Do you see any damage inside and around the USB port? Enlarged USB port? Dry or pulled soldering?
 
An aspect in common with issue https://github.com/prusa3d/Prusa-Link-Web/issues/316 is Prusa Link staying open from the browser for some time during the print.
Are you able to reproduce the issue while Prusa Link is not opened on the browser?

On a side note, you can save a _Crash Dump_ on the USB flash drive via _settings > Save Crash Dump_ after the occurrence of BSOD.

Michele Moramarco
Prusa Research

GithubUser99999999, 2023-01-03:
> Please use a different USB flash drive, formatted FAT32.

I thought the same and over the last two weeks I tried another old USB 2.0 drive and a newer (faster) USB 3.0 drive with the same results. I also tried no USB extension cord and a second/different extension cord. It always happened again. I also checked all three thumb drives and they are fine.

> Do you see any damage inside and around the USB port? Enlarged USB port? Dry or pulled soldering?

No, it looks normal but it's a tiny bit wobbly from the start and USB devices are a bit loose. 

> Are you able to reproduce the issue while Prusa Link is not opened on the browser?

I also tried closing the PrusaLink browser tab and it still crashed after a random time. I then tried printing for several days with Ethernet and WiFi disabled and had no problems at all, which is why it's network-related and not USB-related. 

To be more precise: I am now only uploading the gcode file via LAN or WLAN with PrusaSlicer and then I switch off network in the printer and then prints work flawlessly as before. So the gcode itself is probably not corrupted by uploading it that way. I didn't try uploading via PrusaLink again so far however.

> On a side note, you can save a _Crash Dump_ on the USB flash drive via _settings > Save Crash Dump_ after the occurrence of BSOD.

Does it also work several days after the last crash? I don't necessarily want to waste more filament by deliberately trying now.

Prusa-Support, 2023-01-19:
Thanks for the information.

The Crash Dump should be automatically generated after a fatal error. I should try to find more information. Not sure how long it is stored but it may be obsolete by now. Normally, we request the user to reproduce the error and exported it immediately after.

So the printer prints flawlessly as long as the Network connection is turned off via the printer menu, right?

At this point, I wouldn't be concerned by a slightly wobbling USB port or by the flash drive because the uploaded print files are saved on the USB flash drive. The flash drive is actively used during the print and I seem to understand that it doesn't cause issues as long as the Network connection is off in your case.

Michele Moramarco
Prusa Research

GithubUser99999999, 2023-01-19:
> So the printer prints flawlessly as long as the Network connection is turned off via the printer menu, right?

Exactly, since I wasted so much filament already, I am currently always printing only with any network functionality disabled like before 4.4.0

Prusa-Support, 2023-01-26:
OK, thanks for the feedback, and please keep it coming as much as you can.
We are taking notes.

Michele Moramarco
Prusa Research

GithubUser99999999, 2023-01-26:
I have updated to 4.4.1 and so far no problems with enabled network, but I also didn't print anything large that takes very long yet.

closed 2024-06-19 by github-actions[bot]