prusa3d/Prusa-Firmware-Buddy #2840: [BUG] Force reset required if USB removed on print preview

hamdeli, 2022-12-18:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type**
MINI+

**Printer firmware version**
4.4.0

**Original or Custom firmware** - [Original]

**Bootloader version**
2.0.2

**Optional upgrades**
Filament Runout Sensor

**USB drive or USB/Octoprint**
USB

**Describe the bug**
If USB is removed whilst the printer is displaying print preview, navigation is blocked. You cannot navigate to "Back" and you are forced to reset to resume any printer function.

**How to reproduce**
1. Either select print or plug in a USB for it to auto select the newest GCODE file to print. 
2. USB state R/W, Preview screen is displaying
   1.  note that on this screen you can navigate back or print.
4. Remove USB
5. Printer deadlocked forcing you to reset.

**Expected behavior**
User should be able to navigate "Back" or the printer should kick back to the "home" screen automatically when USB is no longer detected.

baervevo, 2023-01-05:
Having a similar issue.

Prusa-Support, 2023-01-28:
Thanks for reporting but I'm not able to reproduce the issue on the most recent FW version.

If the one-click print is displayed, I can remove the USB flash drive and I can return to the main screen. Same if I remove the flash drive at any point while browsing and previewing the prints on the USB flash drive.
Most importantly, if the USB flash drive is removed mid-print, the print is paused but can resume after, as usual.
I hope I'm not missing anything.

Please try [FW 4.4.1 (final)](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1).

Michele Moramarco
Prusa Research

Prusa-Support, 2023-02-18:
Is this issue still relevant?
Please try again with [FW 4.4.1 (final)](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1).

In case of a lack of feedback and non-reproducibility, the issue may get closed as stale.

Michele Moramarco
Prusa Research

hamdeli, 2023-02-18:
Closing this issue as resolved.
It is no longer reproducible in [FW 4.4.1 (final)](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1).
When you remove the USB on print preview, the screen force returns to the main menu.

closed 2023-02-18 by hamdeli