prusa3d/Prusa-Firmware-Buddy #2842: [BUG] Printer stuck (reboot needed) when paused before bed leveling during heating up

Eldenroot, 2022-12-19:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - [4.4.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor+Wifi module+Bondtech IFS]

**USB drive or USB/Octoprint**
  USB - uploaded via PrusaSlicer (ethernet connection, sliced and sent via ethernet)

**Describe the bug**
  Send your sliced file to the printer, start print (via PrusaSlicer option), wait few seconds and during the heating phase (before bed leveling phase) and click on "Pause" button. When you click on pause, printer is paused but you cannot return back to printing. All is stucked, you cannot even move between buttons via knob. Only reset can help here (button next to the knob).

**How to reproduce**
  Check above.

**Expected behavior**
  Resume button should be displayed, reboot should be responding, no reboot needed!

**G-code**
  ---

**Crash dump file**
  ---

**Video**
  ---

MistrKernnunos, 2023-01-11:
Hi, this is expected behavior, beacuse pause waits for current gcode to finish and after it is finish the printer is in paused state. 
When you pause the printer during heating the printer is waiting for the heating gcode to finish.
I agree that the printer seeems like it is stuck, but in reality it is just waiting for gcode to finish.

Eldenroot, 2023-01-11:
Hmm but it should display at least some note to user. It is confusing

Prusa-Support, 2023-01-28:
Thanks for your feedback. 
This is not a bug and I'm not sure if it can be improved but I'd leave it open for now.

The printer doesn't seem to behave like that during traditional printing via the LCD panel.
Do you confirm that this only happens if the print is launched via Prusa Link?

Michele Moramarco
Prusa Research

Eldenroot, 2023-01-28:
Sent by PrusaSlicer.

Prusa-Support, 2023-02-09:
That could be considered pretty much equivalent to starting the print via Prusa Link.
Can you reproduce it by starting a print directly via the LCD menu?

Michele Moramarco
Prusa Research

closed 2024-06-19 by github-actions[bot]