prusa3d/Prusa-Firmware-Buddy #2862: [BUG] Error 12602 after flashing 4.4.0, can't downgrade

Rav333, 2022-12-28:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI +]

**Printer firmware version** - [4.4.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint**
USB flash drive
**Describe the bug**
Used Prusa USB stick for 4.4.0 firmware update, two .bbl files on USB. Flashing started. After the first boot an error 12608 shows up (Firmware corrupted). At this point, BootLoader is at 2.0.2 version.
While trying to downgrade or upgrade with 4.4.0. RC1 there goes error 12602:
"USB drive not connected! Please insert a USB drive with a valid firmware file."

Tried with another USB and formatted both to FAT32 again. Nothing seems to change.

**How to reproduce**
Don't know

**Expected behavior**
Enable firmware further upgrades

[image]
[image]

Rav333, 2022-12-29:
Used yet another USB, fresh format and firmware got updated with 4.4.1. RC.
Not sure if above helped or the fact that I turned printer on first and then inserted USB.
Previous FW updates were fine with USB plugged in before turning MINI on.

Prusa-Support, 2023-01-31:
Thanks for reporting.

Initially, the bootloader was not designed to be upgraded like a regular firmware installation. The developers worked hard to allow a simple and seamless bootloader and firmware installation, but quite erratic errors were noticed during the installation process. It may require some polishing, and they already made the installation of FW 4.4.1 much faster and more efficient compared to 4.4.0.

The problem highlighted by the error is potentially true, but I wouldn't exclude it is an erratic error.
Please follow the indication on the error-related article in the [Help](https://help.prusa3d.com/) section. If that doesn't help, I suggest the following troubleshooting steps.

.1
I can speculate that the previously recommended methods to force an installation (`X-button click + knob long press` or `X-button click + knob double-click`) sometimes could cause launching the installation multiple times, which may turn out to be problematic on a heavy installation like for FW 4.4.0.

Please try to force the installation as follows.
1. press once the reset button (press all the way, don't hold)
2. count roughly 3 seconds
3. press once the knob button (press all the way, don't hold)

.2
Try installing the firmware again, one file at a time.
Find a working USB flash drive and format FAT32.
Upload only the smaller .bbf file on it.
Try installing the firmware again, "force the installation" if nothing happens.
Once done, remove the USB flash drive, delete the .bbf file, upload the other bigger .bbf file only, then try to continue the installation, "force the installation" if nothing happens.
This may sound a bit confusing, so you may want to repeat it in case of problems.

.
As far as I can tell, the installation of FW 4.4.1 is much more straightforward so please allow me to close this issue.
Please share feedback and, in case of further problems, we may consider reopening the issue.

Michele Moramarco
Prusa Research

closed 2023-01-31 by Prusa-Support

EddiOS42, 2023-07-09:
I'm having this issue with my mini+. Currently on 4.4.0 trying to update to 4.4.1. Doesn't automatically recognize the new firmware, so I tried the above force installation. Stills shows 12602 error.

Prusa-Support, 2023-09-30:
Please try more than once, but only the small bootloader .bbf file to begin with.

Use a different USB flash drive too.

Sometimes glitches persist unless a factory reset is performed before and after the firmware installation.
Please try a factory reset before the firmware installation.

Also please consider reaching out to Customer Support before adding further reports on this thread.

Michele Moramarco
Prusa Research