prusa3d/Prusa-Firmware-Buddy #2864: Issues forwarding the Prusa Mini+ IP using it outside of LAN connection

FrederikVigen, 2023-01-01:
I have now bought the Prusa Mini+ ESP module, and everything seems to work completely fine on local connection.
Is it possible to allow external communication? 
When forwarding the IP on my router to the Printer, it does not seem to accept any incoming HTTP connections, is there a way to modify the firewall settings on the printer?

mix579, 2023-01-02:
Did you open the port for https? Port 443?

FrederikVigen, 2023-01-05:
Hello Mix yes i tried both of them, and neither are working for forwarding

murk-sy, 2023-01-10:
Since you are connecting via WLAN, it is possible that your router is blocking external access to wirelessly connected devices.

Can you verify with perhaps a laptop that port forwarding to WLAN devices works? I don't believe there's any restrictions to access the printer beyond the API key.

CuriousTimo, 2023-01-10:
It works when I test it.

- Port forward from WAN IP TCP 8080-> LAN IP Pusa Mini TCP 80
- In the browser go to: http://WAN IP:8080 -> Shows Prusa Link

Could the issue be that you are trying to use HTTPS instead of HTTP?
Because HTTPS will result in ERR_CONNECTION_REFUSED. Even on the local LAN

Prusa-Support, 2023-02-21:
Thanks for reporting the issue
Also, thanks to all of the intervened users for such great pieces of advice.

Unfortunately, this behavior is most likely router-specific. It would be difficult to find a universal fix for all router models, especially because the manufacturer or the network provider can apply very specific parameters and limitations.

You may want to reach out to the network provider or the router manufacturer for help solving this.
Once this is figured out, it may be useful to post information about the router model and the solution for future reference.

Michele Moramarco
Prusa Research

closed 2024-06-19 by github-actions[bot]