prusa3d/Prusa-Firmware-Buddy #2878: [ENHANCEMENT] Display material, print time, weight and length of filament

TojikCZ, 2023-01-04:
[image]
Display the gcode metadata from the gcode file on the web interface.
Moved from here: https://github.com/prusa3d/Prusa-Link/issues/612

Prusa-Support, 2023-01-18:
Possibly related to https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2334.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-10-28:
Prusa Connect is up and running! 🎉 
With the new firmware now available in alpha version (potentially unstable), the printer Prusa MINI is enabled on Prusa Connect too.

Prusa Connect, at the current non-final stage, already includes this and more data from the printer and the print files.
Prusa Link, as of now, is expected to remain the most basic possible interface, as it is hosted on the printer's board, and it acts as a bridge between the printer and Prusa Connect.

[image]

Please allow me to close this issue as solved but feel free to share feedback below.

Michele Moramarco
Prusa Research

closed 2023-10-28 by Prusa-Support

jhaand, 2024-04-27:
Prusa Connect is not a solution for locally monitored Prusa printers.