prusa3d/Prusa-Firmware-Buddy #2888: [ENHANCEMENT] Option to show relative time remaining when network connected (or 12 hour time like is standard in my country)

barrettj, 2023-01-05:
I'd much prefer to see the relative time or the time in my country's native 12 hour format.

It's actually quite annoying to choose between having network functionality and easily seeing how much time remains in a print and I'm not sure why this is the default.

murk-sy, 2023-01-06:
+1 also because DST is currently not handled automatically (which would probably require either getting timezone info from Prusa Connect or using up half the firmware space with all the timezone garbage)

Eldenroot, 2023-01-07:
Must have, many similar issue are already here

Prusa-Support, 2023-02-01:
Thank you for reaching us, I should close this issue as a duplicate though.

Please follow the existing issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/648.

Michele Moramarco
Prusa Research

closed 2023-02-01 by Prusa-Support