prusa3d/Prusa-Firmware-Buddy #2898: [BUG] Prusa Slicer - send g-code and start print - HTTP 409 conflict error

Eldenroot, 2023-01-08:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - [4.4.1RC1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Wifi module, IFS Bondtech Extruder]

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  Send file over Prusa Slicer works fine for the first use (printer turn on). Gcode is uploaded and the print starts automatically. However when you wanna print another model (and you click "Home" on display to go to the main screen) and send another Gcode via Prusa Slicer -> all is uploaded, model thumbnail is displayed on the printer LCD screen but the print not start. You have to manually start it. I use Win 11, connected over ethernet (directly PC<->printer). I have tested also via Wifi module and also with another PC. Same issue. I have Win11 Pro, Prusa Slicer 2.5. 

This issue is fixed when you turn off and on your printer again.

**How to reproduce**
  Check above.

**Expected behavior**
  You should be able to send and start printing via Prusa slicer without any error message.

**G-code**
  Happens with any gcode sliced file.

**Crash dump file**
  ---

**Video**
[image]

murk-sy, 2023-01-10:
Duplicate of #2686

1) Can you reproduce it every time after rebooting and one print?
2) Is the correct file always displayed after 409 conflict error?
3) If you have an unused USB drive available, can you format it and try to reproduce the issue to rule out the USB drive?

I am having more or less similar issues, but I am suspecting it's related to the USB drive on my end. If possible, please do a test of the above.

daver360, 2023-01-13:
I'm having similiar issues when I'm trying to start a second print.
I'm still using the original USB stick provided by Prusa. I didn't even reformated so far. But trying a different one is something I'll try.

Eldenroot, 2023-01-15:
@Prusa-Support - always... it happens only when you choose "Send and print"... you can always just send a g-code to printer without any error.

4.4.1_RC2 and PS 2.5

Eldenroot, 2023-01-15:
Here is another issue - same as above - no upload... it works inly when you click upload; if you select upload and print -> this error -> no file on usb flash drive.

[image]

Prusa-Support, 2023-02-03:
Thanks for reporting.

The description of the issue is rather specific, but this may fall into those generic errors that often generate a conflict, being discussed in issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2686.

The fact that turning off the printer temporarily fixes the problem (the first "upload & print" works) makes me think that the problem could be related to two known issues. Namely, the Wi-Fi ESP module could fail auto-connecting and resetting.

The [FW final release 4.4.1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1) supposedly solves such issues. Please give it a try.

There could be other generic and erratic issues vaguely mentioned in issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2686, but the most recent FW release may rule out the known issues mentioned in this issue, and possibly close it.

Michele Moramarco
Prusa Research

Eldenroot, 2023-02-03:
Happens with 4.4.1 too, so still valid

Eldenroot, 2023-02-05:
OK guys, some observations:

- if you set "preheat" option on your printer and you wanna upload and start print -> this issue can be reproduce. Anyway this is only one part and one way how to reproduce. It happens also without preheat enabled, but it is not so often.

Prusa-Support, 2023-02-15:
Preheating doesn't generate a conflict in my case.

I was seldom able to create conflicts like yours, but there was pretty much always a known reason behind it. For example:
`Error 409` when the printer is not on supported screens;
`Error 28` when the printer is busy (e.g. printing).

I've tested the worst possible connection conditions to intentionally interfere with the Wi-Fi signal too (e.g. placing the printer as far as possible from the router + metal and other objects on the way). What I've noticed is that the connection symbol on the printer can take quite some time to refresh and grey out, or may not grey out even if the connection is objectively poor.

In other words, I may think of various suboptimal conditions to generate erratic problems, bound to external factors. 
I've reason to believe that the connection reliability will be naturally improved in various ways on the firmware side in the future, but I'm afraid we are looking at too generic, varying, and erratic problems to be worked on, like the ones mentioned in #2686.

I would suggest closing this issue in favor of issue #2686, where unspecified connection problems are being reported.
Maybe other users active there can reproduce your specific scenario of conflict too.

Michele Moramarco
Prusa Research

closed 2024-06-19 by github-actions[bot]