prusa3d/Prusa-Firmware-Buddy #2912: [BUG] <after stopped Octopi-print the printer wants to cool down>

funtomaas, 2023-01-10:
**Printer type** - MINI+

**Printer firmware version** - 4.4.0

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor (not working)

**USB drive or USB/Octoprint** - USB/Octoprint

**Describe the bug**
  After stopping a print (for example because the nozzle seems stuck) I try to use the clean the Filament option, but doesn't work, cooling down seems dominant. EVEN after reboot by tipping the reset-button!

**How to reproduce**
  1. start printing via Octopi ( I didn't try without octopi)
  2. wait for bed leveling, heating up and first filament extrusion for printing the object on heatbed
  3. then stop printing on Mini+-Screen and leave octopi-connection
  4. try Option: "Clean Filament" (I use german Version : "filament reinigen"). 

**Expected behavior**
  With firmware Version before I could clean the nozzle without delay and restart printing via octopi. But with this Firmware Version the cleaning task is ignored, the temp-display blinks blue and only wants to cool down...

**G-code**
  every Gcode!

**Crash dump file**
  -

**Video**
  -

funtomaas, 2023-01-22:
now it seems clear to me: "purge Filament" in the menu doesn't work at all.

AndiDittrich, 2023-01-29:
same here with `v4.4.1` (menu item is disabled) - additionally "low temp" is displayed in some cases when trying to purge the filament manually via "move e axis" (temp set to `220°` manually and sensor value is steady)

Prusa-Support, 2023-02-25:
Please allow me to close this issue as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2835.

Michele Moramarco
Prusa Research

closed 2023-02-25 by Prusa-Support