prusa3d/Prusa-Firmware-Buddy #2933: [BUG] Duplicate items showing in print menu

reedy, 2023-01-13:
**Printer type** - MINI

**Printer firmware version** - 4.4.0

**Original or Custom firmware** - [Original/Custom]

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint** - USB

**Describe the bug**
I've noticed while trying to print from the files list on a USB stick, it is common for items to show multiple times in the list (after some scrolling), this includes the "Default" folder for example.

Clicking into one of the item, and then returning seems to fix it, but it's happening repeatedly.

Will try and get a video

**How to reproduce**
Try to print from a USB stick, and items will show more than once...

**Expected behavior**
Not have duplicate items appear

**G-code**
N/A

**Crash dump file**
N/A

**Video**
TBC

Eldenroot, 2023-01-15:
Hmm, but it has different names?

reedy, 2023-01-15:
Yes there's a PLA and a PETG file, which is two. But it shouldn't appear three times

reedy, 2023-01-15:
[image]

[image]

[image]

reedy, 2023-01-15:
The Klem_buis is also being shown twice, but there is only one as per the usb drive screenshot...

Eldenroot, 2023-01-17:
Try to use different USB thumb drive and format it.

Prusa-Support, 2023-02-21:
Thanks for reporting.

I'm sorry to say we didn't hear of similar problems and I'm unable to reproduce it.

There are many similar files and the only difference is the end part of the name. I don't want that this causes a conflict with sorting and listing files, but I don't see how that can happen either.
Also maybe there is a third similarly named file incorrectly sorted at the beginning. Maybe we can't see other similarly named files in the top part of the screen capture from the computer, because the folder is sorted by _name_ but _last modified_.
I wonder if you can reproduce it on a new flash drive, where file names are unique, simple, and shot (like alphanumeric codes A1, A2... B1... C1...), so that the whole name will be shown on the small printer's display.

That's surely an interesting anomaly but please, try to use a different USB flash drive (up to 16 GB or 32 GB) formatted FAT32.
If the problem will persist, please also perform a factory reset on the printer.
Moreover, have you tried to upgrade or downgrade the firmware?

Michele Moramarco
Prusa Research

reedy, 2023-02-22:
>Maybe we can't see other similarly named files in the top part of the screen capture from the computer, because the folder is sorted by name but last modified.

In this case, you've got non deterministic behaviour. It shouldn't be there and then disappear, it would be always there.

But the folder isn't sorted by name on the printer, it's sorted in the same way as in Finder. The most recently added files are shown first (which makes sense for most workflows; if you've added a file, you probably want to print it.

I've had quite a few issues with the Prusa issued memory stick (it's very slow on my macbook to write files, for example. doesn't like to eject either), but I have replaced it with a new 32GB sandisk drive... I've not tried formatting or otherwise the original provided USB stick...

I've never downgraded the firmware, but have since upgraded to 4.4.1. The new memory stick arrived January 14, and I've been using it since.

Prusa-Support, 2023-03-04:
Yes, the items should not randomly appear and disappear, but some kind of corruption in the _last modified_ metadata could make them be listed incorrectly. Without evidence, this remains pure speculation though.
Sorry I didn't mean to say that sorting by _last modified_ method is preferable, but, in addition, sorting by _name_ would be useful to check if there are other files similarly named.

Is it safe to tell that all of the problems were not due to a firmware bug but due to a corrupted USB flash drive?

Michele Moramarco
Prusa Research

Prusa-Support, 2023-03-23:
In summary, the problem doesn't seem related to the printer's firmware but to the memory stick. 

As a side note, according to https://github.com/prusa3d/Prusa-Firmware/issues/4089#issuecomment-1471449175, files created or modified within 3 seconds would be sorted out erratically as an inherent limitation of FAT filesystems.
We can't exclude that some files were created/modified at once, almost simultaneously.

The issue will be closed as not reproducible and most likely not relevant to the firmware.

Michele Moramarco
Prusa Research

closed 2023-03-23 by Prusa-Support