prusa3d/Prusa-Firmware-Buddy #2938: [BUG] Possible USB wiping issue with 4.4.1?

baaguette, 2023-01-17:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.
I am sorry if this issue
**Printer type** - [MINI]
Mini+

**Printer firmware version** - [e.g. 4.0.5, ...]
4.4.1+4194

**Original or Custom firmware** - [Original/Custom]
Original
**Optional upgrades** - [e.g. Filament Runout Sensor]
Prusa filament sensor 
**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint
USB flash drive (the cool one that comes in the box)
**Describe the bug**
  A clear and concise description of what the bug is.
  
  I am first and furmost sorry for the lack of precision or clear evidence on this, I haven't considered encountering a bug since upgrading on the newest version and haven't properly logged my progress throughout the day.
  
Considering I am but an amateur in the field, I cannot quite fully catch what exactly happend but in short after I have failed a print because of a probable lack of rigor while cleaning the nozzle. I have proceded to clean it again and then do some basic calibration via the software to be sure of everything to be a-1 before restarting again my progress.
(Mesh bed leveling, selftest, first layer calibration if i remember correctly)
When went ahead to retry my print , Ive found my usb key with only one file "project" with nothing else inside. If I do happen to look inside, there is nothing but a folder with "..". There is then a small loop if i try to click on the knob a couple of times and then back to the main menu.  There was normally originally around 20-30 gcodes files ready to go counting the basic ones and the ones that i am currently trying out and messing around or a raspberry pi project.
[image]
[image]
[image]

So afterwards, I went on my computer (Windows 10 desktop) to check inside the key to see if the files were still there on it or not. To then see a error message telling me that there need to be a formating done on the keydrive (G:\ in this case) because the content inside may have been corrupted. (images to follow- I will roughly translate what it is said but tldr it is windows basic option to say that there is an issue with the drive)

[image]
[image]
(Screenshot 1 -  G: is unaccessable - The volume doesn't have any known system files. Verify if all the system pilots of the files are loaded or if the volume isn't damaged)
(Screenshot 3 -  You need to format the disc in the G reader before using it - do you wish to format it?)

Truly I do wish to be able to provide with more information, but I quite do not know what would be useful information no more about this topic. I truly wish not to waste your time with this, considering I do have backups files on my computer and it is more an inconvinience then anything else. I only mean to try to help out in case it causes further issues to someone else.

**How to reproduce**
  Please describe steps to reproduce the behavior.

Only god knows at this point. 
Sorry to not be able to provide with more informaiton. 
Possibly try out to calibrate a machine with the tests said above on the new 4.4.1 version after stopping a print around 10% completed?

**Expected behavior**
  A clear and concise description of what you expected to happen.
  I have not noticed any flicker or anything of the sort, I have also not manipulated the usb drive from it's slot during the cleaning so i don't quite know further then this...

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.
  
  I have no longer access to any log error files considering they were once in that usb key.

**Video**
  Please attach a video. It usually helps to solve the problem.

Eldenroot, 2023-01-18:
Actually this happenes to me as well.

Thumb drive was empty, I had to format the drive. I have lost all files. It happened with 4.4.0 beta fw.

I do not how to reproduce, I thought it is because of the broken thumb drive but now I am reading your report...

baaguette, 2023-01-18:
I'm sorry to hear that It happened to you aswell, but I am also reasured that I'm not going crazy!

Prusa-Support, 2023-01-26:
Hello! Thank you for your report and sorry for the issue. From the description, it looks like something went wrong with the USB drive, indeed the drive became "corrupted". I'd recommend checking if formatting that drive in FAT32 might resolve it - or if another working, FAT32-formatted USB drive would regularly work on the printer. We'll keep an eye if there might be more cases related with 4.4.1 specifically, though we don't suspect at the moment a relationship with the firmware version. 

Let us know, in case of anything to review, do not hesitate to ask! 

Alessandro Pantaleo
Prusa Research

baaguette, 2023-01-26:
Thank you so much for your time. Indeed just reformating the usb drive do fix it so if someone ever happens to have this issue, try it out!

closed 2023-01-26 by baaguette