prusa3d/Prusa-Firmware-Buddy #2941: [ENHANCEMENT] Additional status: End of printing (also in API)

eWuPi, 2023-01-18:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - [MINI]
**Optional upgrades** - [e.g. Filament Runout Sensor]

**Is your suggested improvement related to an existing problem? Please describe.**
The print completion status is Idle

**Describe the expected functionality**
After the printout is completed, the printer sets the End of printout status (minimum 1 min) for some time or until the temperature of the bed or nozzle is reached and the fans are turned off. It is useful in Home Assistant to undertake certain automations after the printout is completed, e.g. Changing the status from End of Printing to Idle to Turn off the power.

Prusa-Support, 2023-02-22:
Thanks for reaching us, I'm not sure I understand the problem/suggestion though.

I may seem to understand that you are interested in automated actions on the printer at the end of the print. That may be possible by entering Prusa-specific code in the "custom end G-code" box of Prusa Slicer.
https://help.prusa3d.com/article/prusa-specific-g-codes_112173

Michele Moramarco
Prusa Research

eWuPi, 2023-02-23:
I want the printer to display the "End of printing" status, not just "Waiting", because the waiting status appears in various situations and I would like to have a certain message, e.g. to use the API via Home Assistant to send information to the phone, that the printout is over.
PL: Chodzi mi o to, żeby drukarka wyświetlała status "Koniec wydruku", a nie tylko "Oczekiwanie", ponieważ status oczekiwanie pojawia się w różnych sytuacjach a chciałbym miec pewny komunikat np. po to aby korzystajac w API poprzez Home Assistant wysłać informację na telefon, że już koniec wydruku.

Prusa-Support, 2023-03-07:
It may be possible in the future via Prusa Connect and Telegram.
https://forum.prusa3d.com/forum/postid/632290/

I hope you're happy with the answer. 🙂 

Michele Moramarco
Prusa Research

Prusa-Support, 2023-04-19:
This issue may be closed as invalid.
Please clarify the report/request, or close the issue if your doubts have been removed.

Michele Moramarco
Prusa Research