prusa3d/Prusa-Firmware-Buddy #2943: [BUG] Upload from Prusaslicer fails, upload via Prusalink succceeds

mix579, 2023-01-19:
I sent two sliced files from Prusaslicer, no problems. Then with the third one (after info box said transfer of second one was complete) I get the error: Error uploading to print host HTTP 409: Already in transfer process. Only one file at time can be transferred.

Tried several times over 10 minutes, no luck.

Then exported the G-code file and uploaded it via PrusaLink web interface, no problem at all. Back to Prusaslicer, tried again, same error (even though I was hoping for a "file already exists" error…

**Printer type** - MiNI

**Printer firmware version** - 4.4.1

**Original or Custom firmware** - Original

**Optional upgrades** - IFS, Dragon

**USB drive or USB/Octoprint**
  USB flash drive and PrusaLink

TheMoen, 2023-01-27:
For clearity, i am using a wired connection. Prusa Mini+ - firmware 4.4.1 - Revo micro upgrade

I too had some issues with uploading files from prusaslicer. And it seemd like the printer had about 70% uptime. 
Attempting to troubleshoot the issue i noticed that the printer still had a network connection while the "adapter" setting was set to None. when it was set to none, the site was showing and it did not  allow logging in. Then i tried to set the adapter to "wi-fi" still only connecting by RJ45 cable. 

I am assuming you have a wired connection too?
I am curious if there is a bug with the selection of adapter.
I hope prusa can fix this. In the meanwhile attempt setting wifi as the adapter while still using lan??

Logs from home assistant, while testing this out.
[image]

Prusa-Support, 2023-02-25:
Thanks for reporting.

I'm afraid we are looking at yet another sporadical issue due to an unstable connection,
 pretty much another hardly reproducible generic connection failure like the ones reported in issues https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2686, https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2772, and https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2898.
 
I may be missing something but I couldn't reproduce it. Please suggests the steps for a possibly accurate reproducibility.

My advice is to keep on posting these generic and unreproducible connection issues in one place (it can be the first issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2686), closing the other similar issues.

Michele Moramarco
Prusa Research

Eldenroot, 2023-02-25:
It is not about connection... all is about the printer itself - when you upload and print from Prusa Slicer - all is fine for the first time (after power on the printer)... anyway when you finish, you wanna upload and print other model from Prusaslicer and you leave the creen on "print finish" or click during upload and go to the main menu it happens!

mix579, 2023-02-25:
I think it's fine to put this under one issue #. But I'm with Eldenroot—at this point it seems pretty well established to me that at least some of the issues reported are not connection issues.  Especially #2686 feels like a clear case of the firmware detecting a new file on the USB stick and throwing up the preview screen blocking the print command from Prusalink. Assuming that it only checks periodically for new files that would explain why sometimes it's faster than the Prusalink print command and sometimes it's not, in which case the uploaded file will be printed fine

Prusa-Support, 2023-03-07:
There could be various known and unknown reasons for a print upload to fail.
As of now, a key factor to confirm the mentioned generic problems (one, a few, or all) is reproducibility.
Sooner or later, we may need to close a few of these generic issues about problems that can't be controllably reproduced.

The firmware and the connection stability are expected to be improved release after release.
We will keep improving the firmware, _also_ thanks to your reports.

Michele Moramarco
Prusa Research

jagr200, 2023-03-18:
> There could be various known and unknown reasons for a print upload to fail. As of now, a key factor to confirm the mentioned generic problems (one, a few, or all) is reproducibility. Sooner or [...]

I could reproduce this 100,000 times consecutively. Why is that so hard for Prusa to understand? If I power up the printer. Then immediately proceed to upload a print it will work and start printing. If I then clean off the bed and upload a different print to start and the printer is still on the "finish print" screen then it will fail the upload. That is NOT a connection issue. It is poor programing. When a print is officially done change the screen back to the main menu. Pretty simple task. This is 100% on the prusa developers to fix and should be an easy change. I can manually go back to the menu and then print. But that should not be necessary. Its an annoying task that I have to remember to do. Especially when most printers are away from people's workstations.

TheMoen, 2023-03-18:
It might be that prusa is not with you that is stays on the "print finished" view.
I got the same(?)issue
When it is on the "print finished" i cannot "upload and print"

Prusa-Support, 2023-03-30:
> If I then clean off the bed and upload a different print to start and the printer is still on the "finish print" screen then it will fail the upload

Prusa Link won't work correctly on the "finish print" screen at the moment.
We are aware of that and we have a separate issue about it (https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2301).
Our developers will work on it.

This was also implied by the final release notes (November 2022).

> *MINI+ must be on the home screen and “print-preview” screen - this is intentional for safety reasons

https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0

If _"finish print" screen_ is the cause of all troubles, this issue can be closed.

Michele Moramarco
Prusa Research

milladekilla, 2023-05-26:
Soo just upgraded my mini with Bondtech feeder and this issue started .to consistently appear.. since i see there is IFS upgrade in the original post i think this might be connected as there are some changes done in "developer" hidden section of HW settings, perhaps it has some influence on API access as what i get in console is :

in.2db555ec7e6e55de710c.js:1     POST http://10.0.4.200/api/files/local 404 (Not Found)
(anonymous) @ main.2db555ec7e6e55de710c.js:1
y @ main.2db555ec7e6e55de710c.js:1
M @ main.2db555ec7e6e55de710c.js:1
n.onchange @ main.2db555ec7e6e55de710c.js:1
main.2db555ec7e6e55de710c.js:1     POST http://10.0.4.200/api/files/usb 404 (Not Found)

The1Percent, 2023-05-27:
There seems to be limited error checking or connection recovery capability on this. I am dropping zero packets to the printer. Why is this not a more robust function at this point? This is a solved problem in the world no?

Prusa-Support, 2023-07-18:
We are on this but so far the reported issues are quite erratic. 
I'm positive that the natural firmware optimizations will improve the data transfer and the connectivity features in general.
For now, we would like to focus on the controllably reproducible issues and the most specific conditions for the issue to manifest.

Michele Moramarco
Prusa Research

closed 2024-06-19 by github-actions[bot]