prusa3d/Prusa-Firmware-Buddy #2945: [BUG] Mini+ hotend overheating error on 4.4.1

fifal, 2023-01-20:
**Printer type** - [MINI]

**Printer firmware version** - [4.4.1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor, wifi module]

**USB drive or USB/Octoprint**
USB drive through Prusa Link

**Describe the bug**
I wasn't able to print anything after flashing the new 4.4.1 firmware as I was getting the hotend overheating message (https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/src/lang/po/cs/Prusa-Firmware-Buddy_cs.po#L755) everytime I tried to print something. 

After downgrading to 4.4.0, everything works as usual.

**How to reproduce**
I was installing [ai-thinker](https://www.laskakit.cz/ai-thinker-esp-01s-esp8266-wifi-modul/) wifi module, so I downloaded the 4.4.1 on the USB drive and followed the steps in the documentation on how to install the wifi module. The installation was successful. The next day when I tried to print the attached G-code I was getting the hotend overheating message.

**G-code**
[USB plug](https://data.filek.cz/USB_plug_0.2mm_PLA_MINI_19m.gcode)

mix579, 2023-01-21:
Do you get the message before the print starts or during the print. My workaround for this new "feature" that no one has asked for is to preheat the nozzle manually to 170 degrees before starting a print, which for me has worked so far.

fifal, 2023-01-22:
I got this message before the print started right after the mesh bed leveling. Shortly after I have written this issue I have checked the official manual on how to replace the hotend thermistor. In one of the picture I noticed that tip of the thermistor in my heaterblock did not reach the end like in the picture:

[image]

I decided to try to push the thermistor inside the heaterblock so it was aligned with the edge. Then flashed the 4.4.1 firmware again and this time it worked flawlessly.

I'm not really sure that this was the actual problem as on the 4.4.0 firmware it was working correctly, but it works for now.

Grimthul, 2023-01-27:
Had the same problem as @fifal and solved it the same way (pushed thermistor to the correct position). Thank you, you saved me a lot of time.

AMPERAGE0, 2023-02-01:
I had the same issue. I'd get the error either just as the print was starting, or within about 2 minutes of it commencing.

Pre-heating the hot end didn't help. I had to roll back to 4.4.0.

Prusa-Support, 2023-02-25:
Thank you for reporting.

@fifal. Yes, the thermistor position is relevant, as highlighted in the issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2768.

Please allow me to close this issue as a duplicate.

Michele Moramarco
Prusa Research

closed 2023-02-25 by Prusa-Support