prusa3d/Prusa-Firmware-Buddy #2948: [BUG] <Live Z value returning 1.0>

seifer69er, 2023-01-21:
**Printer type** - [MINI]

**Printer firmware version** - [4.4.1]

**Original or Custom firmware** - [Original]

**Describe the bug**
  A clear and concise description of what the bug is.During first layer print, the live z value in the footer returns 1.0
This value does not change unitl >1.0mm

i.e at layer;
0.2, it will return 1.0
0.4, it will return 1.0
0.6, it will return 1.0
0.8, it will return 1.0
1.0, it will return 1.0
1.2, it will return 1.2
1.4, it will return 1.4

**How to reproduce**
Checked with both z axis options in the footer
Z axis and Live Z axis, both behave the same.

**Expected behavior**
0.2, it will return 0.2
0.4, it will return 0.4
0.6, it will return 0.6
0.8, it will return 0.8
1.0, it will return 1.0
1.2, it will return 1.2
1.4, it will return 1.4

This is the excerpt from 
footer_item_live_z.cpp in src 4.4.1

string_view_utf8 FooterItemLiveZ::static_makeView(int value) {
    static std::array<char, 7> buff; //"-2" - "0", longest "-0.001"
    int value_clamped = std::clamp(value, -2000, 0);

    int printed_chars = snprintf(buff.data(), buff.size(), "%i.%.3u", value_clamped / 1000, std::abs(value_clamped) % 1000);

    if (printed_chars < 1) {
        buff[0] = '\0';
    } else {
        //dont want it to erase last in 0.0, -1.0, -2.0
        while ((--printed_chars) > 2 && buff[printed_chars] == '0' && buff[printed_chars - 1] != '.') {
            buff[printed_chars] = '\0';
        }
    }

thijstriemstra, 2023-01-23:
how in the hell does a bug like this get introduced?! I am astonished that **so many** issues have been introduced since 4.4.0 and it looks like it keeps getting worse. Please Prusa, introduce some unit tests or revert everything introduced after 4.3.4 because this is snowballing into a disaster :( really dissapointed, I bought a prusa because of the great firmware updates everyone was talking about but this is out of control.

Eldenroot, 2023-01-23:
There are many regressions… and unfortunately the system how they lead this repository (tickets) is not a good example. They should clean up all tickets here, many of them are invalid. Also, changelogs are not complete, so you do not know what they break or fix.

Prusa-Support, 2023-02-22:
Thanks for reporting.

We are tireless workers in a young company, running an ever (fast) growing business.
Surely there is room to improve on all sides and we believe that our community feedback is most valuable.

Thank you again for your support.

Michele Moramarco
Prusa Research

seifer69er, 2023-02-24:
love to help where I can, background in Electronic and Hardware Engineering :)

closed 2024-06-19 by github-actions[bot]