prusa3d/Prusa-Firmware-Buddy #2956: [ENHANCEMENT] <Enter comprehensive title>

krupkaj, 2023-01-25:
Please, could you add info about WIFI module firmware version into to screen Info -> System Info?

Prusa-Support, 2023-02-28:
Thanks for your suggestion.

Our developers will look into this.

Michele Moramarco
Prusa Research

bkerler, 2024-02-09:
I just wrote a PR for this, enjoy :)

CZDanol, 2024-02-13:
Internal ticket BFW-5075

CZDanol, 2024-02-19:
Hello, after consultation within the company, we've decided what we will not implement this.
* The ESP FW number is basically an internal number that doesn't have much significance to the user.
* Each printer FW has its own required ESP firmware and it will not work properly with any other.
* In the upcoming release, there will be a different UI element added showing the status of the Wi-Fi module, including the "Wrong version" (or something along those lines) status.

closed 2024-02-19 by CZDanol