prusa3d/Prusa-Firmware-Buddy #2962: [QUESTION] Extrusion multiplier from PrusaSlicer

plandevida, 2023-01-30:
Please, before you create a new question, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Leave your comments or questions here.**

Hey!

I have been playing for a while with TPU and struggling a lot. On the way I noticed that changing the extrusion multiplier in PrusaSlicer does not change the printer settings, the flow settings remains at 95 always.

Is this supposed to be like this or it should reflect that?

Cheers,
Daniel.

Eldenroot, 2023-01-30:
You can change it in your printer gcode.

plandevida, 2023-01-30:
For somebody without gcode knowledge it does not feel like how it should work.

is it meant to be like this or is it a bug?

plandevida, 2023-01-30:
Could you tell me the gcode instruction? So I can check if the issue is in PrusaSlicer or in the firmware

Prusa-Support, 2023-03-01:
Thanks for asking.

I confirm this is intentional behavior, not concerning in any way, and to be expected on all printers.
It is so for legacy reasons.

Fun fact.
You can notice that on MK2 and MK3 series too.

I hope this answers your question and that this issue can be closed.

Michele Moramarco
Prusa Research

plandevida, 2023-03-01:
Thanks Michele for the clarification.

This issue can be closed since there is no bug.

Anyway I think is a topic to consider, generally I would expect people expecting the same behavior as I did. At the end the settings is in PrusaSlicer and I understand it does not have any effect.

closed 2023-03-01 by plandevida

Prusa-Support, 2023-03-01:
Yes, `extrusion multiplier` is in _Prusa Slicer_ and the `flow` in the _printer's tuning menu_ are not directly related.

Michele Moramarco
Prusa Research