prusa3d/Prusa-Firmware-Buddy #2967: [BUG] Wired Ethernet port does not respond to Ping

jweaver999, 2023-01-31:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - Mini+

**Printer firmware version** - 4.4.1

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint** - USB Drive

**Describe the bug**

Wired Ethernet port does not respond to a ping

**How to reproduce**

Open a cmd tool (Windows) and try to pIng the address.. you will get 100% loss

λ ping 192.168.1.75

Pinging 192.168.1.75 with 32 bytes of data:
Request timed out.
Request timed out.
Request timed out.
Request timed out.

Ping statistics for 192.168.1.75:
    Packets: Sent = 4, Received = 0, Lost = 4 (100% loss),

**Expected behavior**

The Ethernet interface should respond to the ping

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

crabbedh, 2023-01-31:
If you look at the link:  https://e2e.ti.com/support/microcontrollers/msp-low-power-microcontrollers-group/msp430/f/msp-low-power-microcontroller-forum/782195/msp432e401y-icmp-ping-checksum-is-set-to-0

You’ll see some software requires you select an option necessary or the PING reply has zero checksum. Likely this was not done in the firmware and a ping reply was never tested.

ZD4ve, 2023-01-31:
The screenshot from my forum post:
[image]

Prusa-Support, 2023-03-01:
Thanks for reporting.

Our developers will look into this.

Michele Moramarco
Prusa Research

schwabe, 2023-09-28:
I can confirm this issue on the Prusa XL and the checksums of all 0. It is only present when using Ethernet. The macOS ping utility also ignores the invalid pings.

JohnOCFII, 2023-09-29:
Also can confirm this remains a problem with MK4 firmware 5.0.0.  Behavior is different on the wireless interface.  This improperly formed ICMP packet is on the wired Ethernet interface.  I've attached some example packet captures in WireShark PCAP format for both successful ICMP pings (to a different device) and the failing ICMP pings to the MK4.

[ICMP-pcap.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12754748/ICMP-pcap.zip)

schwabe, 2024-05-27:
It looks like the issue is fixed on the latest firmware for my XL

jweaver999, 2024-06-07:
I just checked my Mini which is running 5.1.0 and it appears that its fixed

λ ping 192.168.1.15

Pinging 192.168.1.15 with 32 bytes of data:
Reply from 192.168.1.15: bytes=32 time=1ms TTL=255
Reply from 192.168.1.15: bytes=32 time<1ms TTL=255
Reply from 192.168.1.15: bytes=32 time=1ms TTL=255
Reply from 192.168.1.15: bytes=32 time=3ms TTL=255

Ping statistics for 192.168.1.15:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 0ms, Maximum = 3ms, Average = 1ms

So i would say this can be closed.

stixpjr, 2024-06-07:
Also confirmed fixed on my Mini running 6.0.1:

```
ksh$ ping -c 5 prusa
PING prusa.private (192.168.0.91): 56 data bytes
64 bytes from 192.168.0.91: icmp_seq=0 ttl=255 time=0.727058 ms
64 bytes from 192.168.0.91: icmp_seq=1 ttl=255 time=0.817601 ms
64 bytes from 192.168.0.91: icmp_seq=2 ttl=255 time=0.718387 ms
64 bytes from 192.168.0.91: icmp_seq=3 ttl=255 time=0.782615 ms
64 bytes from 192.168.0.91: icmp_seq=4 ttl=255 time=0.750209 ms

----prusa.private PING Statistics----
5 packets transmitted, 5 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 0.718387/0.759174/0.817601/0.041044 ms
```

closed 2024-06-08 by danopernis