prusa3d/Prusa-Firmware-Buddy #2970: Filament sensor issue when changing colors

redheadmaker, 2023-02-03:
Created a print with Prusa slicer that has a color change. Running through Octoprint. Printer stops for color change and extrudes filament - asks if it was extruded - answer yes - then it says Please remove filament from filament sensor - there is no filament in sensor, screen just is stuck on that message - no buttons for o.k. or anything. This has happened twice in a row. I guess I will see if I can turn off filament sensor. I do a lot of 2 color prints for items I sell on Etsy, so this is a problem for me.

Prusa-Support, 2023-02-07:
Hello! Thank you for the report. As a first check, what would happen if you attempted reproducing this, but without Octoprint, disconnecting it temporarily and running the print through USB drive? Would there be a similar situation or anything different? 

Alessandro Pantaleo
Prusa Research

ShadowDrakken, 2023-02-24:
I just had this same issue occur printing from PrusaLink on my MINI+

Prints up to the first color change, ejects the filament without issue. Then refuses to believe the filament sensor is clear. Had to reboot the machine to get out of the menu, then tested the sensor and the sensor is working just fine during a manual load action.

4.4.1 Firmware

ShadowDrakken, 2023-02-24:
Update on this issue: running the same print from the LCD menus, the same error appeared to clear the filament sensor. However, when I pushed filament in and back out, the error cleared itself, and I was able to continue the print. The same is NOT true of starting the print from PrusaLink :(

pboguslawski, 2023-04-23:
4.4.1 Firmware same problem wish printing stuck on remove filament from sensor message when no filament is present. Only cold reset was possible.

Prusa-Support, 2023-06-07:
Thanks for your feedback.

I seem to understand that the issue only occurs when the print is started remotely (via Octoprint or Prusa Link).
I wonder if the problem could be related to starting a print from remote or to the network connection itself.
Does it actually make a difference if the print is started from remote, or if the printer is simply connected to the network?

Have any of you been able to reproduce 100% of the time when the print is started from remote?
Have anyone been able to reproduce the issue when printing directly via USD drive?

Removing and reinserting the filament seemed to help on one occasion.
Have you happened to try that even when connected/printing from remote?
Please avoid fast and frequent re-inserting but wait a few seconds between actions.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-06-07:
On a side note, the issue may be related to or a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1448.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-06-13:
> However, when I pushed filament in and back out, the error cleared itself, and I was able to continue the print.

@ShadowDrakken Do re-inserting filament work as a walkaround in either case (remote print or print via LCD menu)?

Was the printer still connected to the network while printing via LCD menu?

Michele Moramarco
Prusa Research

ShadowDrakken, 2023-06-13:
> @ShadowDrakken Do re-inserting filament work as a walkaround in either case (remote print or print via LCD menu)?
Eventually. It usually takes removing and reinserting several times

> Was the printer still connected to the network while printing via LCD menu?
Initially no, but the problem persisted in either case

brobertson, 2023-06-23:
I've just had this same issue.

ShadowDrakken, 2023-06-27:
Given that reverting back to the pre-Wifi firmware resolves the issue, I'm fairly confident it's not an issue with the sensor having debris in it.

sjors-lemniscap, 2023-07-29:
MK4 Firmware 4.7.1 here and can confirm this is an issue (although slightly different scenario). My issue is when the filament ran out and I had to feed another spool of filament. All original Prusa products (I'm using PLA).

Reinserting the filament didn't help and I had to manually pull out the remaining filament from the extruder. The flow was as follows:

1. Ran out of filament and my print stopped automatically
2. Confirmed that I want to change filament
3. Confirmed that I want to purge remaining filament
4. The issue starts here, when it asks me whether all filament is out and I say "no" there is no option to purge out more filament. When I normally change the filament (not mid-print) it actually allows me to purge more filament
5. Because the filament has not been purged enough there is still some quite some stuck in the extruder (see attached picture)
6. Manually removing it is an option BUT since it's mid-print and quite some force has to be used to remove the filament from within the extruder there is a high change you are moving the extruder and can't continue your print.

It would be great to have the option to purge more filament while replacing the filament mid-print (just like how this is possible when changing the filament outside of a print).
[image]

bpaccount1, 2023-08-08:
I just have the same problem.

MK4 is stuck on:
"Changing filament
0%
Please remove filament from filament sensor."

There is no prompt to continue. After successfully removing filament and trying to load more, nothing changes on the screen and it will not pull in the new filament.

There's nothing blocking the filament hole or sensor.

I'm on PrusaLink via WiFi connection and MK4 firmware alpha4.

ErwanAliasr1, 2023-08-12:
I went on the same issue with Firmware 4.7.2 on my brand-new MK4.
I had to insert and remove the filament 10 times before it offered me to insert the filament for loading.
The printing resumed properly after that.
Sounds like a bad calibration/state of the sensor.

7s3, 2023-08-12:
> I went on the same issue with Firmware 4.7.2 on my brand-new MK4.
> I had to insert and remove the filament 10 times before it offered me to insert the filament for loading.
> The printing resumed properly after that.
> Sounds like a bad calibration/state of the sensor.

I couldn’t get it to trigger no matter how many times I reinserted the filament.

ErwanAliasr1, 2023-08-17:
@Prusa-Support Could we get a feedback on this ?
This thread was started 6 months ago and proved various products are affected the same way and could make some prints void.

FliesWithWind, 2023-08-26:
+1 on this. It just happened to me when I runed out of filament. Wasted 24h of print....

Prusa-Support, 2023-08-26:
We acknowledge that more users are reporting this issue, we won't ignore that.
However, we are still unable to reproduce the issue.

So far, as far as I can tell, pretty much all cases directly handed by us were related to debris or faults of various nature.
In other words we can't confirm a bug yet.

**🔊 To all Prusa MK4/XL owners in the thread. 🔊 
While the firmware shares part of the MINI's base code, MK4/XL firmware is substantially different.
This issue was originally created for Prusa MINI (and opened before the MK4/XL release).
Please consider posting a separate issue - _only after consulting our official [Customer Support](https://help.prusa3d.com/article/customer-support_2287)_.**

Michele Moramarco
Prusa Research

ErwanAliasr1, 2023-08-26:
If you can't reproduce it and consider that's a debris issue, it would be
nice to let your customers having workarounds when this occurs. Having a
menu to retry the filament eject could probably help many of us by trying
moving this debris in the head. That would avoid us loosing hours of print.
WDYT ?

sjors-lemniscap, 2023-08-27:
> We acknowledge that more users are reporting this issue, we won't ignore that. However, we are still unable to reproduce the issue.
>
> So far, as far as I can tell, pretty much all cases directly [...]

What I don't understand is why you simply don't trigger the "Change Filament" flow as how it would be triggered manually and where it gives us the option to "Purge more" filament so that whatever is "stuck" can be purged out.

It's interesting that Prusa is unable to reproduce this bug. Myself and a friend of mine own both a Prusa MK4, when we simply cut the filament and let it run out mid print the issue occurs as described in my previous post. Funny enough when my print failed because the filament ran out and the sensor doesn't detect my newly inserted filament, and I trigger the "Change Filament" flow manually from the menu the sensor doesn't have any issues at all.

DevNull8, 2023-08-30:
I was having the same problem with my MK4 and I appear to know what's causing the bug. I have the same problem when the filament runs out or if I try to do a mid print color change the printer gets stuck on Changing Filament 0% message. Looking through the posts here I noticed anyone who seems to have the network options enabled are experiencing the issue. I have PrusaLink and Prusa Connect enabled although I don't really use them. On a hunch I disabled both of them and tried my color change print that failed. The printer now successfully gets past the changing filament message and finishes the print.

It would appear there is some sort of bug that the printer will not sense a new filament being loaded if the PrusaLink or Pursa Connect network settings are enabled. Someone else who is having the problems try disabling Prusa Connect or PrusaLink and see if the issue goes away for you! If I feel like it I may try enabling each one individually to see which one is causing the problem, but at this point I feel like if someone else can replicate then Prusa should have enough to look into this bug.

DZDee, 2023-09-23:
Add me to the list of people with this problem. I was printing from the USB however I do have WiFi enabled. I just starting have problems with connect today. Anyway doing a two color print and was successful with the first. The next two have had the hang on remove filament 0% bug. I turned off the filament sensor to try to get these prints done. On both failed prints I did open the quick disconnect door and verified there was no leftover filament. Both times it was clear. Sensor or software was faulty. This is a big issue that needs attention from Prusa support. Not too pleased with such a simple function not working and wasting filament. Also the online functionality is really confusing not very features and unnecessarily complex. I know we are nerds and all but come on this could be so much better.

biting-midge, 2023-11-04:
I have 1h23minutes print time - I ran out the test sample of filament today and have exactly the problem described above.

I have both Prusa Link and Connect enabled.  The filament path is clear, there is no problem with the sensor as far as I can see as the filament runout was detected perfectly and the remaining few cms of filament was ejected successfully.

The only exit for me was to reset.

duneking1, 2023-12-26:
So I’m having this same issue with MK4 latest firmware as of Dec. 25th.

run out of filament and the print stops.  Prompts you to remove the filament.  I find if I remove the filament before addressing what is on the display the printer doesn’t recognize that I’ve inserted or removed any filament.  Shoving it in and out does not progress the screen at all.  In the past I had to reset and throw away a lot of prints.  

Today I had luck for the first time and opened the sensor door and shoved the print in across the wheel.  It didn’t work for the first several attempts but eventually I got the beep that progressed the screen.  

This seems to happen because I just reach in and take the filament out before addressing, and following, the instructions on the LCD.  

This is very likely a bug and I can easily replicate it.  

There needs to be prompt on the display that allows you to override the sensory and just tell it to feed the new filament.

paulcobbaut, 2024-01-01:
I just had the same issue. After 8 hours of printing there is a scheduled colour change and the mk4 display says "please remove filament from filament sensor" and there is no other option. I tried at least twenty times to insert new filament, but it did not work. I cut the power and turned it back on, but am still unable to insert filament. I think there is filament stuck inside so will need to open the nextruder (first have to get the printer out of the Prusa enclosure).

RedsGT, 2024-01-16:
I had this same issue and I THINK what happened is when the initial filament sensor calibration was done, I pushed the filament all the way in instead of slowing inserting it and immediately stopping once it was detected.  Running the calibration again from the settings and doing it correctly this time (as stated by the on screen directions) fixed the issue for me.

Prusa-Support, 2024-02-02:
Debris may cause filament sensor malfunctioning on pretty much _all filament sensors_ we tested and implemented so far.

The reproducibility of this issue is mostly due to the presence of debris, except for unusual cases of deformed parts, hardware faults, and wiring problems.
These are not the only variables though. For example unexpected user interactions and external interferences... Not to mention applications like connections to third-party boards or to the network, that are not thoroughly tested (yet) - e.g. https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2822, https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3111.

_Is there a bug? Not confirmed to date._ 🤔 
_What is the cause then? Hard to tell, but we have evidence that often it is debris (mixed with other rare scenarios)._
_Is there room for improvement? "Only" always_ 🙂 _but you may want to check for debris._ 

The implementation of the MMU on MK4, will be our best chance to further polish the retraction process and drastically reduce the chances of the debris inside the Nextruder (see https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3129) - in a similar fashion to its implementation on MK2 and MK3.
This enforces the idea that this issue has been spread on all printed models, with different hardware and firmware.

For now, please mind inserting and removing the filament a few times to possibly knock off debris, or access the filament sensor for inspection and cleaning.
https://help.prusa3d.com/article/ir-filament-sensor-troubleshooting-mini-mini_230431

Also, please help us distinguish between **MINI** and MK4/XL cases (because hardware and settings may differ), and the cases related to serial line or network connections: please check other relevant issues like the ones linked above before posting.

Michele Moramarco
Prusa Research

DanPow21, 2024-02-03:
I wanted to say that I just encountered this exact same issue. I ran out of filament 7 hours into an 8 hour print, the printer said to unload the filament, I did, and it remained on the screen that says:

"Changing filament
0%
Please remove filament from filament sensor."

I tried reinserting filament to no avail. Then I found this page and read the whole thing. Then I just repeatedly rammed the new filament in and out of the filament sensor quickly for 20 seconds and it eventually beeped and progressed to the next screen. 

So I had a happy ending. I guess I'm glad it ended up being a "simple" fix. My print continued and finished successfully with the new filament.

martona, 2024-03-04:
This keeps happening to me when running out of filament mid-print, prompting a filament change from the LCD, or doing a sliced-in filament change operation in the gcode.  It doesn't seem like a hardware issue with the sensor (dirt, etc.) as the printer is still very new (MK4) and it detects running out of filament every time; what it doesn't detect is the subsequent manual removal of the ejection stub.  This is what happens:

1) Trigger filament change (run out naturally during a print, cut the filament mid-print with scissors, use the LCD to prompt for a change, etc.) - to me the sensor works here every time, I have deliberately cut the filament several times to make sure...

2) Extruder goes to a parking position and ejects the little bit of filament stub it still has inside, prompts to remove filament from the sensor.

3) Following this prompt and removing the stub is not acknowledged and the printer is stuck here until reset or power cycled.

For people ending up here with this problem: you can continue your print if you insert your filament (push it in gently as far as it goes) then power cycle. Keep your hand on the filament, gently pushing down. When the printer reboots it will resume the print job as if it had just experienced a power loss. The extruder will start moving left and right, but it's running on empty. Only when you feel that the filament is moving dowbwards between your fingers should you let go. Do not use the reset button as that counts as a user abort and will not resume the print. You have to reach behind the printer for the power button.

martona, 2024-03-22:
I figured out what might be a problem in a large number of cases. This is definitely a software issue.

When a filament change (either due to a change gcode or runout detection) happens, these are the steps that normally occur:

1) Printer parks head
2) Printer ejects filament or filament stub
3) A question is displayed: "Is filament ejection successful?"
4) User answers yes
5) Printer says "Please remove filament from filament sensor"
6) User removes filament
7) Printer says "Please insert filament..." (paraphrasing) - from here on the process is normal and works fine.

If the user follows these steps in order, everything works fine.   However, if the user performs step 6 before step 4, the printer softlocks. That is, if you do not answer the filament ejection question in step 4, but pull out the stub without being prompted to do so, and THEN answer the question in the affirmative, the message in step 5 will be displayed, and nothing you do will get the printer to detect a filament state change. 

This is definitely a problem and 100% reproducible on my MK4.

Prusa-Support, 2024-03-24:
> 3. A question is displayed: "Is filament ejection successful?"

The user can't possibly answer that question without even trying to remove the filament.
The user should remove the filament already and, if not successful, the printer will offer new options accordingly.
The printer won't request the user to remove the filament if the filament was removed and the filament sensor was disengaged.

> 5. Printer says "Please remove filament from filament sensor"

Probably the printer asks so because the filament sensor is still triggered. Otherwise, it won't insist on this but simply wait for the user to load new filament.

Cutting power will abort this operation and the Power Panic will recover the print, resuming from the last gcode line stored in the memory, in fact ignoring the filament change operation and the status of the filament sensor too.

_Please add the filament sensor status footer to monitor the status of the filament sensor reading in case of doubts._

We keep monitoring the situation, collecting relevant feedback, and studying the case. However, _so far_, the only _confirmed_ scenario is that the filament sensor ball & spring are stuck onto something.
**The recommended steps remain the following.**
- Remove and insert the filament multiple times in an attempt to knock off debris and realign the filament sensor ball. 
- Blow canned air from [the hole above the filament sensor](https://cdn.help.prusa3d.com/wp-content/uploads/2023/08/5ac2eee63621ff1e9030242fe03fa27c.jpg).
- Blow canned air from below, through [the opened idler door](https://cdn.help.prusa3d.com/wp-content/uploads/2023/03/XL_Extruder_Open2.jpg).
- Remove the Planetary Gear parts "[like a sandwich](https://cdn.help.prusa3d.com/wp-content/uploads/2023/10/67c593556671a61a79c52036477d3454.jpg)" to prevent gear slippage. Hence access, remove, clean, and realign filament sensor ball & spring. Repeating the [Gearbox alignment](https://help.prusa3d.com/article/gearbox-alignment-mk4_465779) is very recommended after this operation.
- In the case of a recurrent issue please consider
  - drying filament or comparing the recurrence of the problem with other filaments,
  - repeating the Filament sensor calibration ([_Calibrations & tests_](https://help.prusa3d.com/article/lcd-menu-mk4_413171) menu).

Basic guidelines can be found in the relevant Help article linked below.
https://help.prusa3d.com/article/filament-sensor-mk4-mk3-9-xl_465702

Michele Moramarco
Prusa Research

CZDanol, 2024-05-23:
Closing this in favor of #3129

closed 2024-05-23 by CZDanol