prusa3d/Prusa-Firmware-Buddy #2972: [BUG] Divide By 0 blue screen when OctoPrint Pi power is unplugged

handro123, 2023-02-05:
**Printer type** - Original MINI (not +)

**Printer firmware version** - 4.4.1 

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor, SuperPINDA, E3d Revo Micro

**USB drive or USB/Octoprint**
Have OctoPrint on Raspberry Pi 4 installed and Connected by not printing. 

**Describe the bug**
My OctoPrint was not responding so I went to reboot the Raspberry Pi while still Connected to the Printer. When I unplugged the power to the RPi it also reset the Printer and the printer showed a blue screen for "Divide By 0" error. First noticed the issue on 4.4.0 firmware, then upgraded to 4.4.1 and was still able to reproduce the issue. 

**How to reproduce**
1. Have printer online and Connected to OctoPrint.
2. Unplug power from Raspberry Pi while OctoPrint is Connected to printer.
3. Printer will reboot and show blue screen for "Divide By 0" error.

**Expected behavior**
I don't think the Printer should reboot and show blue screen if the Raspberry Pi loses power.

**G-code**
N/A 

**Crash dump file**
Can't attach .bin so it is inside Zip archive. 
[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/10614132/dump.zip)

**Video**
Issue on 4.4.1: https://www.youtube.com/shorts/1cf9c34nFHE
Issue on 4.4.0: https://www.youtube.com/shorts/exlSn63JKjU
[image]
[image]

Prusa-Support, 2023-03-10:
Thank you for reporting.

That's surely interesting, and I don't think I've ever heard about this specific message before.

Have you tried further firmware downgrading?
Would the error persist on firmware 4.3.4 (or older)?
Could you confirm that the error is "something new" (4.4.0 or newer firmware)?

Michele Moramarco
Prusa Research

handro123, 2023-03-11:
Downgraded to 4.3.4 and the issue was not seen. 

Upgraded back to 4.4.1, had to go through the Setup Wizard again, and after that was successful I was able to reproduce the "Divide by 0" error again by disconnecting power to the Raspberry Pi running Octoprint. 

[image]

Prusa-Support, 2023-04-11:
Alright. Thank you for the additional feedback.
Our developers will look into this.

Michele Moramarco
Prusa Research

ulululu12, 2023-04-13:
Just to confirm this.
Same issue occured for me.

blindguynar, 2023-07-10:
I ran into this blue screen today for the first time.  I wasn't unplugging octoprint, but my pi is powered by poe and was running for a few days now as i was printing heavily.  

Normally i only turn on/off to print.   i rebooted prusa and pi and within a short time (10 min) i had the blue screen again.  I shut down the pi to give it a rest and ran across open issue this looking for a solution.    i am on firmware 4.4.1+4194 which is what came on my mini.

Just adding my hat to the ring of those it happened to since this issue is still marked as open.

Prusa-Support, 2023-09-30:
Thanks for reporting.

So far an abrupt power shortage on the micro USB port seems to be the key to the reproducibility of the issues as further confirmed by other users in a more recent issue (https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3278).

Michele Moramarco
Prusa Research

closed 2024-06-19 by github-actions[bot]