prusa3d/Prusa-Firmware-Buddy #2980: [BUG] Wifi does not work correctly on Prusa MINI after installing module for PrusaLink

joshtgreen, 2023-02-07:
**Printer type** - MINI
**Printer firmware version** - 4.1.1

**Original or Custom firmware** - Original
**Optional upgrades** - Filament Sensor

**USB drive or USB/Octoprint**
Neither, bug is with installing a wifi module and using PrusaLink.

**Describe the bug**
I bought and installed [this wifi module](https://www.amazon.com/gp/product/B00O34AGSU/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&psc=1) (from the tested list) as outlined in [these instructions](https://help.prusa3d.com/guide/how-to-setup-wi-fi-mini-mini_316781). Everything went fine, and I got a "connected" message at the end of the process and the Wifi icon is colored white on the printer. However, the IP address is still listed as 0.0.0.0. I first reseated the module, re-flashed the firmware, and checked the credentials. 
 
The issue persisted so I logged into my router and found the device. The router had assigned the device an IP address. I tried to connect to the printer in PrusaSlicer using the IP address from the router, but got the following error:

"Could not connect to PrusaLink: Timeout was reached: Connection timeout after 10000ms [Error 28]"

**How to reproduce**
 See above bug description. 

**Expected behavior**
I expected the assigned IP address to display on the printer and for the PrusaSlicer to be able to connect the printer via PrusaLink. 

**Crash dump file**
Attached. 

[dump.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/10678886/dump.bin.zip)

Prusa-Support, 2023-03-11:
Thanks for reporting.
Firmware version 4.4.1, isn't it? I think 4.1.1 is a typo. 

Please double-check the Network settings in the settings menu, and make sure the selected interface is _Wi-Fi_.
You can surely also test the Ethernet connection for troubleshooting purposes.

On a side note, this may be a duplicate of issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2303. If not, we may have not received similar reports before and we are not able to reproduce it on our own. In that case the issue may not be confirmed.

Please notice one DIYmall model in the list was not approved - also it was different from the product picture according to the notes, so there is a chance it was a poor-quality clone.
Just please don't exclude the possibility that you received a faulty unit or a clone.
It may be useful to test a different Wi-Fi module or reach out to the vendor to get a replacement.

Before considering that, please try a factory reset and a new firmware installation.
Also, this may sound obvious but, please try again after moving the printer closer to the modem and make absolutely sure you are connected to a 2.4 GHz Wi-Fi signal.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-04-13:
This issue will be closed as a seemingly duplicate of the existing issue mentioned above, and stale.

Michele Moramarco
Prusa Research

closed 2023-04-13 by Prusa-Support