prusa3d/Prusa-Firmware-Buddy #2986: [BUG] Failed filament change during print

TheMoen, 2023-02-12:
**Printer type** - Prusa Mini +
**Printer firmware version** - 4.4.1 (official)
**Original or Custom firmware** - Original
**Optional upgrades** - Filament Runout Sensor and Revo micro hotend
**USB drive or USB/Octoprint**  Usb drive and prusalink

**Describe the bug**
 During a 8 hour long print the second "filament change" failed, I did not watch the printer so it was about an hour cold before i got to change the fiilament.
- After removing the filament, i pressed the button confirming i unloaded it
- Manually pushed in filament into the extruder gear(at this point i was suspicios since the printer did not grab onto a cm as it usually does)
- The printer hotend heated up.
- The printer thought it had filament and started printing.

- I quicly paused the print. attempted to fix it while the printer was paused(not possible by the way)
- Pulled out filament from the filamentsensor
- Started the print, expecting the printer would stop since the filamentsensor was "false".
- The printer starter printing
- opened the menu pressed change filament.
- Changed the filament successfully
- Continued print sucessfully.

**How to reproduce**
Not shure if it is possible to reproduce, but attempt trying to wait 1,5 hours after the printer starts the the secound filament change during the print.

**Expected behavior**
Change filament successfully
Stop the print when filament sensor is "false"
Enchancement: allow to access settings while paused

Prusa-Support, 2023-03-11:
Thanks for reaching us.

Changing filament after a few hours of waiting is a recurrent scenario in our office 🙂 and while it is not ideal, it doesn't necessarily cause the filament change to fail.
Also, a `is color correct?` question should show up, giving you the option to retry.
It seems that the failure is not reproducible.

If the filament sensor is "false" (filament runout) the print will be paused indeed.
Unfortunately, the filament sensor won't be triggered if the filament is not grabbed, because the filament sensor is mounted before the extruder and can't detect filament motion.
If you have a filament sensor, you can still remove and re-insert the ungrabbed filament to trigger a filament outage (during the print) and a consequent filament change.

During the print, you have access to some settings in the _tune_ menu. There you can find an option to change filament.
If you paused the print, you can resume and open "tune" or even open the idler door to attempt a manual filament loading (not recommended).

I may be missing something so please double-check and provide pictures and video documentation about the issue if that can help explain the problem.
For the time being I hope you don't mind me closing this issue as invalid.

Michele Moramarco
Prusa Research

closed 2023-03-11 by Prusa-Support

TheMoen, 2023-03-11:
This has only happened once, the printer skipped both loading and confirming that it was the correct color.
Probably wont happen again, so i guess the issue is neglishable.