prusa3d/Prusa-Firmware-Buddy #2987: Wifi is utterly unstable

rusty-labs, 2023-02-13:
**Printer type** - Prusa mini

**Printer firmware version** - [4.4.1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**Describe the bug**
90% of the time I'm unable to upload gcode by wifi with PrusaSlicer.
After a few percent of progress I get 
[image]
Also I'm able to open prusa link front page, but it shows N/A to everything
[image]

I've to mention that If I print anything from USB flash drive then I have a much higher chance to print via WiFi afterwards, and prusalink webpage starts to show valid status

Connection to wifi module is 59dBm.

**How to reproduce**
  Send a gcode to a printer by Wifi

swalberg, 2023-02-20:
I get the same error uploading from Prusa Slicer, though I've only tried maybe 5 times and haven't seen it work. After I've tried that and it fails, the web interface is broken the same way as in the original ticket.

However, if I start the printer up and connect to the web interface, everything works as expected. I can log in, upload files, and print.

So, this works:
* Power on printer
* Log in to web interface
* Upload file
* print

This doesn't work:
* Send gcode to printer from Prusa Slicer 2.5.0+MacOS-64 by wifi
* Watch it fail (not immediately, it'll upload for a bit)
* Try to go to the web interface, see that I get the same thing as the original ticket

plandevida, 2023-03-21:
Hi,

I am having the same issues, I cannot send GCODES from PrusaSlicer, I got errors all the time.

The most common error is 408 (Time out), I manage to print 1 from 15 tries.

On the other hand, upload to print is very slow, when the file is uploaded on the USB already the printer takes around 1-2 minutes to start printing.

The WIF connection is also appearing and disappearing all the time, but the set up and the test connection from PrusaSlicer is always good and flawless.

Pog3k, 2023-04-03:
Same here.
After reataching the USB Sticks it works. 
Hope this hint may help.

Prusa-Support, 2023-05-02:
Thanks for reporting.

We may be looking at yet another sporadic (or too generic) connection failure like the ones reported in issues https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2686, https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2772, https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2898. 

I may be missing something but I couldn't find the steps to controllably reproduce the problem.
Our developers and collaborators need to reproduce the specific problematic scenario to be able to analyze it.

This issue will be closed as a duplicate.
You can take part in the ongoing discussions on the aforementioned issues.

Michele Moramarco
Prusa Research

closed 2023-05-02 by Prusa-Support