prusa3d/Prusa-Firmware-Buddy #2997: [BUG] Delayed extrusion after travel when using 0.6 mm nozzle

rasssta, 2023-02-18:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]
Mini

**Printer firmware version** - [e.g. 4.0.5, ...]
4.4.1+4194

**Original or Custom firmware** - [Original/Custom]
Original

**Optional upgrades** - [e.g. Filament Runout Sensor]
Filament runout sensor
0.6 mm nozzle
WiFi

**USB drive or USB/Octoprint**
Octoprint. But I have the same issue when printing from USB.

**Describe the bug**
I have an issue with scars and holes in perimeters, that I believe started to show when I switched to 0.6 mm nozzle. I'm not actually 100% sure it's related to the nozzle since the issue only shows in certain prints, so it's not really clear when it begun. But the problem also seem to appear with 0.6 mm nozzle for the MK*, see https://github.com/prusa3d/Prusa-Firmware/issues/2760. Prusa support has been involved in that ticket and referred me to create a new one in this repo.

The issue is that extrusion is delayed after certain travel moves by the print head, which causes scars and holes in the print. I've tried to slice with PrusaSlicer, SuperSlicer and Cura and I get the same problem with all of them. I've also tested to remove the filament sensor but the problem persists. PTFE tubes have been replaced to new ones.

It seems to be mainly related to extrusions that are made after the print head is moved from the left hand side to the right hand side of the sheet.

In the print example (picture attached) I placed 4 towers, one in each corner of the sheet. The tower on the right was on the bottom right, the top tower was on the top right, the left one on the top left and the lower one on the bottom left.

But if I move the right hand towers close to the middle I still get the same issue with them, whereas if I move the left hand towers to the middle they print just fine. So it's not just about the actual placement of the object on the sheet, but rather the combination of placements and the print moves it creates.

The problem also occurs with single objects where the same type of movement is performed, in the print, from left to right on the sheet.

I have made some videos where I've filmed the print head and extrusion wheel and the extrusion wheel starts moving at the exact same frame as the print head is lowered (after z-hop), for the parts of the print where I get the holes/scars. I haven't compared it to the non-hole objects but I would assume that the extrusion should start when the print head is lowered?

**How to reproduce**
Swap to 0.6 mm nozzle. Put an object in each corner of the sheet which has only 1 perimeter.

**Expected behavior**
No holes or scars in the print.

**G-code**
See next post.

**Crash dump file**
None

**Video**
None. But phots will be attached.

rasssta, 2023-02-18:
[image]

rasssta, 2023-02-18:
Gcodes:

Cura: [OPM_rettest_2.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/10773744/OPM_rettest_2.txt)

PrusaSlicer: [rettest_0.6n_0.2mm_PLA_MINI_15m.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/10773743/rettest_0.6n_0.2mm_PLA_MINI_15m.txt)

rasssta, 2023-02-19:
I just changed back to a 0.4 mm nozzle and I get the same error so it's not related to the nozzle size

rasssta, 2023-02-20:
I have created two demo/test STLs where each tower is marked with I/II/III/IIII or none (which is center tower, I will call it number 5) in the 5 tower STL.

STLs (rename to .stl):
[Rettest_4tow_0.4mm.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/10785753/Rettest_4tow_0.4mm.txt)
[Rettest_5tow_0.4mm.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/10785754/Rettest_5tow_0.4mm.txt)

4 tower test:
Print order: Tower 3, 4, 2, 1
Tower 4 has a lot of gaps
[image]
A lot of stringing for tower 2
[image]
All other towers are overall okay.

5 tower test:
Print order: Tower 1, 3, 5, 2, 4
A lot of stringing for tower 4
[image]
Gaps are now on tower 5 instead
[image]

rasssta, 2023-02-20:
Reverted to FW 4.3.4+2835. Same issue.

art-0, 2023-02-23:
Your problem is related to retraction length. Each type (PLA, PETG) of filament needs a different one. Change the retraction length e.g. to 1
[image]

also read about linear advance e.g: https://forum.prusa3d.com/forum/hardware-firmware-and-software-help/linear-advance-for-the-mini/#post-618399

rasssta, 2023-02-24:
I'm well aware of both retraction length and linear advance. Both have been extensively tested and tuned without any difference in regards to this problem.

rasssta, 2023-02-28:
Adding another STL with 2 towers that is faster to print:
[Rettest_2tow_0.4mm.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/10851476/Rettest_2tow_0.4mm.txt)

Prusa-Support, 2023-03-04:
Thanks for testing and reporting.

For clarity, it seems that the print in the first picture doesn't match the .gcode file prepared in Prusa Slicer.
The Prusa Slicer G-code was not using pre-settings. The retraction is a bit exaggerated and LA is 0.04 (I would expect it to be around 0.12 or so for PLA + 0.6 nozzle).
I don't want such little things to make the problem look even worse.

I could reproduce the problem in similar print scenarios though.

I recommend focusing on reducing the print complexity so that more users can easily reproduce the same print scenario without worries.

For example:
1. Use Prusa Slicer genuine pre-sets, stock `0.4 mm nozzle` printer, `Generic PLA` filament, `0.2 QUALITY` print profile.
2. Use rounded shapes to reduce speed and flow alteration around the corners.
3. `0% infill` and `1x perimeter` for better visibility of the issue
4. Painted `seam line` towards the center, so that the nozzle won't cross the perimeters during the travels.

[test.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/10888846/test.zip)
**notes**
- objects and perimeters printed counterclockwise, starting with the front-right corner

**result**
- all objects are affected
- the first object (front-right) is most affected _(after travel from-left-to-right)_

1st in the sequence
[image]

2nd in the sequence
[image]

3rd in the sequence
[image]

4th in the sequence
[image]

posen4u, 2023-03-04:
> Thanks for testing and reporting.
> 
> 1. Use Prusa Slicer genuine pre-sets, stock `0.4 mm nozzle` printer, `Generic PLA` filament, `0.2 QUALITY` print profile.
> 2. Use rounded shapes to reduce speed and flow alteration around the corners.
> 3. `0% infill` and `1x perimeter` for better visibility of the issue
> 4. Painted `seam line` towards the center, so that the nozzle won't cross the perimeters during the travels.
> 
> [test.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/10888846/test.zip) **notes**
> ...
> **If more users can reproduce and confirm the issue, that would be appreciated.**
> 
> A _shaking or deformed hotend PTFE tube_ could be a concern! It would create more artifacts like this. Consider cleaning the hotend and reseating the hotend PTFE tube if your results are noticeably worse with the same settings. Before my test in the pictures I did that, just to be on the safe side. 🙂
> 
> Michele Moramarco Prusa Research

I sliced your provided test file on my MK3s, using the `Prusament PLA`, `0.2 QUALITY` and Prusa Slicer default 0.4 printer profile printing with Prusament PLA Blend My Silverness. Note, my printer is running in stealth mode. I tested this twice, first having the model centered on the bed and then move each tower to the corner of my larger printbed.

- **Test 1**, towers centered on bed with the same distances as on the mini bed.

[image]

- **Test 1, All 4 towers** (placed in their representative corners)

[image]
[image]

- **Test 2**, towers moved further out the corners for MK3 size.

[image]

- **Test 2, All 4 towers** (placed in their representative corners)

[image]
[image]

- **Comparison bewteen test 1 and 2**

[image]

**Print files**

- [nocrossing_mk3s.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/10889887/nocrossing_mk3s.zip)

rasssta, 2023-03-05:
> For clarity, it seems that the print in the first picture doesn't match the .gcode file prepared in Prusa Slicer. The Prusa Slicer G-code was not using pre-settings. The retraction is a bit [...]

Regarding the retraction I don't actually recall which settings I used back then as I've tested so much stuff back and forth by now.

But I was actually a bit surprised that 0.12 is the default LA setting when I started to look into LA, and did the linear tuning test print found here https://marlinfw.org/tools/lin_advance/k-factor.html and got the following result:

[image]

0.04-0.05 seems to be the best according to the test.

rasssta, 2023-03-05:
> 1. Use Prusa Slicer genuine pre-sets, stock `0.4 mm nozzle` printer, `Generic PLA` filament, `0.2 QUALITY` print profile.
>
>     2. Use rounded shapes to reduce speed and flow alteration around the [...]

I printed the test1nocrossing.gcode found in the print and got the following result:

1st:
[image]

2nd:
[image]

3rd:
[image]

4th:
[image]

rasssta, 2023-03-05:
> Regarding the retraction I don't actually recall which settings I used back then as I've tested so much stuff back and forth by now.
>
> But I was actually a bit surprised that 0.12 is the default LA [...]

Oh, my bad! I was under the impression that the printed values were multiplied x10 for some reason. Will try with 0.40 instead of 0.04.

rasssta, 2023-03-05:
Another test print, this time with my normal settings (still K0.04 for this one). 99% sure of the print order:

1st (bottom left):
[image]

2nd (top left):
[image]

3rd (top right):
[image]

4th: (bottom right):
[image]

rasssta, 2023-03-05:
Another test, this time with my usual settings (as in the previous post) but this time with K-value of 0.42 instead of 0.04.

Prusa Mini, 0.4mm nozzle, FW 4.3.4+2835, AddNorth PLA Economy White, M900 K0.42.

1st:
[image]

2nd:
[image]

3rd:
[image]

4th:
[image]

rasssta, 2023-03-05:
I have confirmed that the same problem exist in FW 4.2.0+2067 and 4.3.0+2756.

posen4u, 2023-03-05:
> ... MK3s tests...
> [image]

Adding that I am using firmware 3.12.1-5686 on my MK3s. I did another test using a reverse bowden setup as I had it available, and that made no real difference in print quality. However I swapped the filament out for "Clas Ohlson PLA" and ran the same gcode, no issues. So for the MK3 maybe this is only a case of fine tuning the default `Prusament PLA` profile.

**Reverse bowden setup**

[image]

**Reverse bowden result**

[image]

**Clas Ohlson PLA result** (not reverse bowden)

[image]

elHatchling, 2023-03-05:
[image]
I Tried the test.zip on a Prusa Mini+, FW 4.3.4. Bought it March 2021. 
The only mod is that I am using the  "BondTech Heatbreak Prusa Mini".

Prusaslicer 2.5.0, Used 0.20 QUALITY, Prusament PLA profile, 0% infill. More or less everything fromm the .3mf file in the zip.

All cylinders printed fine, with no gaps, but with some stringing.

elHatchling, 2023-03-30:
Regarding the Bondtech Heatbreak. Only thing I did was to use a shorter PTFE tube, according to the instructions from Bondtech. Nothing else have changed. No  firmware parameters or print settings tweaked. 
Using stock Mini Firmwares when updating and using stock Prusaslicer settings for printing.

krcm0209, 2023-04-02:
> New reports of the [test](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2997#issuecomment-1454763541) made with other MINI printers, are welcome.
> Please include pictures and relevant information about hardware, firmware, print settings and filament.

I printed the `test1nocrossing.gcode` supplied by @Prusa-Support on my Mini.

**Modifications from a stock machine**

1. Swapped to SuperPINDA
2. Swapped to Revo Micro hotend

**Versions**

Firmware: 4.4.1+4194
Bootloader: 2.0.2
Buddy Board 1.0.0 (1120X017XC0434)

**Stats**

- X axis: 5.4 km
- Y axis: 4.9 km
- Z axis: 37.1 m
- Filament: 271.6 m
- Print time: 3d 5h

**Photos**

[image]
[image]
[image]
[image]
[image]
[image]
[image]

krcm0209, 2023-04-02:
I can't really back this up with science or evidence (partly because I don't have the time right now to deep dive on this subject), but I have a feeling this behavior is in large part due to:

1. the internal tolerance of the stock bowden tube on the Mini
2. The transition from a very gentle bend radius at the left-most position to a relatively extreme bend radius at the right-most position causing a pulling effect on the filament due to a changing path length

ryanklafuric, 2023-04-02:
Following up on this thread via a reddit [post](https://www.reddit.com/r/prusa3d/comments/129nk2i/help_with_gaps_in_prusa_mini_prints/) in [r/prusa3d](https://www.reddit.com/r/prusa3d/).

I printed the test gcode provided and had some fairly gnarly results. As expected, results were worst in the front right. The best result was the front left. I also suspect that this issue is related to the bowden tube, like @krcm0209 mentioned.

I'm going to try changing filaments, and using a different spool holder to see if that has any positive impact.

Photos of all four:
[image]

[image]

Front right (worst):
[image]

Rear right:
[image]

Rear left:
[image]

Front left (best):
[image]

Asides from this test, I've observed this gap affecting prints like this [Gridfinity](https://www.printables.com/model/242711-gridfinity-master-collection) cube:
[image]

rasssta, 2023-04-03:
> I can't really back this up with science or evidence (partly because I don't have the time right now to deep dive on this subject), but I have a feeling this behavior is in large part due to: [...]

I'm in the same position, can't back it up with science but I have been thinking about just the same two points. I don't know if there actually is a proper solution to this issue (but only workarounds) as it seems to be related to hardware and concept (bowden).

However, I can see 3 possible directions (feel free to comment!):
1. A function in the slicer which compensates for the movement of a bowden printer (can be for Mini or Universal). Let's say you specify where the bowden tube is located on the printer (right, top, top right, etc.) and then the slicer will build some form of compensation mesh for how much of extra restart length or deretraction to use after each travel depending of position A and position B for the travel.
2. Pretty much the same as 1, but implementing this in the Prusa Mini firmware instead of the slicer. Will probably have the same end result, but only solve this issue with the Mini and not for other printers using PrusaSlicer.
3. Redesigned bowden concept for the Mini. Instead of having the tube exit the extruder upwards it might be an idea to move the extruder and redirect the exit direction 0-45 degrees downwards, hence being able to use a shorter tube and more direct angle for the tube towards the printer head. Note that this is just my average Joe brainstorming so take it with a grain of salt :-)

Also, thanks to the "newcomers" in the thread for your contribution with test prints! The Reddit thread highlights that this is an actual problem for people so its nice to have more people giving input here in the issue.

rancor1223, 2023-04-03:
> New reports of the [test](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2997#issuecomment-1454763541) made with other MINI printers, are welcome. Please include pictures and relevant information about hardware, firmware, print settings and filament.
> 
> Michele Moramarco Prusa Research

Here is my result with basically stock Mini+, only with a 0.6mm nozzle. I will repeat this test with 0.4mm in next few days.

**Printer type** - [MINI]
Mini+ (purchased in 2021/10 and used as a hobby printer; the PTFE tube was never replaced)

Printer firmware version - [e.g. 4.0.5, ...]
4.4.1+4194

Original or Custom firmware - [Original/Custom]
Original

Optional upgrades - [e.g. Filament Runout Sensor]
Filament runout sensor
0.6 mm nozzle
levendig silicone sock on the heatblock

To generate the Gcode I used stock settings (for 0.6mm nozzle) in PrusaSlicer 2.5 for "Generic PLA" (I used eSUN PLA+).

[image]

[image] 
(prints are ordered clock-wise starting with bottom left corner of the bed)

----

I'm glad this is finally being looked into, I've struggled with it for a good while (although I don't remember this being a problem back when my printer was new).

d-sko, 2023-04-04:
Hi, I've printed the 3mf file in the test zip as is on a Prusa Mini without any modifications running firmware 4.4.1, here are the pictures:

[image]
[image]
[image]
[image]
[image]

vintagepc, 2023-04-04:
Here are my test results. Mini has had a nozzle change and the base modification but the extrusion system remains stock. 
It's interesting to note there is also underextrusion on the base layers of the circle that is most affected by the gap.

[image]
[image]
[image]
[image]

rasssta, 2023-04-04:
> It's interesting to note there is also underextrusion on the base layers of the circle that is most affected by the gap.

Good catch! Haven't noticed that myself before.

ScreennameTaken, 2023-04-04:
Ok so i've run the test gcode on my mini from the first post (4.4.1, with a Revo and a Bondtech extruder. nozzle used was the 0.4)
The problem only appeared on the last cylinder that the mini comes from left to right.

[image]

https://user-images.githubusercontent.com/24358412/229854349-cdc9fe89-8a8b-47f1-8181-75b9ecbd03fc.mp4

LegsLucas, 2023-04-04:
Prusa Mini+ (1 year old, 7d print time)

**Firmware:** 4.4.1 +4194

**Mods:**
Bondtech IFS extruder
E3D Revo 'Prusa Mini Edition' 
Capricorn XS PTFE tubing
[Revised X End](https://www.printables.com/model/421948-prusa-mini-x-axis-end-idler) 

**File:** test1nocrossing.gcode 
**Filament:** Atomic 'Perfect Orange' PLA
**Nozzle:** Brass 0.4mm

**Results:**

[image]

[image]

[image]

[image]

[image]

ryanklafuric, 2023-04-04:
What I find fascinating is that users with the upgraded Bowden extruder are also experiencing this “under extrusion after travel” issue (apparently). 

I’d be curious if people with a direct drive conversion experience this issue as well.

lemonnfresh, 2023-04-05:
Stock Mini+
.04mm nozzle
4.3.4
Purchased March 2022

Issue: Printer appears to under extrude after travel. I've masked the symptoms by upping the extrusion multiplier to 1.05. Symptom often appears on walls/perimeters on lower right of prints especially those that run directly across the x axis.

[image] All Four, Front View.
[image]Front Closeup

[image]Rear Closeup

[image]All Four Matte, Front View

[image]Front Closeup Matte

[image]Rear Closeup Matte

Lagersuufer, 2023-04-06:
Stock Mini+
0.4 mm Nozzle
Purchased 2020
FW: 4.4.1
SUNLU PETG, Gray

Exact same issue. Results of printing the test file are identical to those already posted, under-extrusion in on the cylinders on the right size after a large X-axis move.

aab010799, 2023-04-06:
Completely stock Prusa Mini+
Purchased February 2023

[image]
[image]
[image]
[image]

antp123, 2023-04-06:
I also have the same problem, firstly noticed it when I switched to 0.6 mm nozzle and printed big part, where seams where set to Random (it left holes in some places)

Printer Type - Prusa Mini+, everything is stock
Printer firmware - 4.4.0
Filament - Prusament PLA Galaxy Silver

Ran the provided test print with:
0.6 nozzle (default PrusaSlicer settings for 0.6mm nozzle)
0.4 nozzle (default PrusaSlicer settings for 0.4mm nozzle)
Then tried to change Bowden tube to a new one (from extruder to hotend) and performed the test print again, but it didn't make much of a difference.
Results are here:
[image]

Also there is difference in wall thickness even at Front Left cylinder:

[image]

aab010799, 2023-04-06:
I should also note that my Mini generally prints the seams randomly regardless of which settings I have. It's quite frustrating. Many of my prints have zits all over despite using the "aligned" settings in Prusaslicer.

slurpdenon, 2023-04-10:
Just wanted to throw in that I've had these issues since I got my Mini+ at the beginning of the year. Talked to support at the time, and they recommended cleaning and reseating the PTFE in the hotend, but that didn't fix anything. Then we assumed it was moist filament that was causing oozing between retractions, but that also didn't resolve my issues. Support seemed out of ideas, so I've just assumed I was doing something wrong, so I'm glad I stumbled across this and now know I'm not crazy. Unfortunately I've wasted a lot of filament after a supports failed because of this on a big print :(. Hope there's a fix for this soon, It's quite frustrating to deal with, and I notice it slightly in nearly every print I've done since I first got the printer.

ncdc, 2023-04-10:
Here are my results with a 0.6 nozzle. Retraction length 2.1mm as that's what I normally run to avoid underextrusion-after-retraction issues.

First
[image]

Second
[image]

Third
[image]

Fourth
[image]

Matto5577, 2023-04-12:
Here is my test results. I have stock mini+, only 0.6 mm brass nozzle. 
Ran the provided test print with:
0.6 nozzle (default PrusaSlicer settings for 0.6mm nozzle)
Printer firmware - 4.4.1
Filament - Azurefilm PLA Black
[image]
[image]
[image]
[image]
[image]
[image]

ScreennameTaken, 2023-04-12:
> FYI, I had this problem exactly. I fixed it by replacing the hotend PTFE and the Extruder to Hotend PTFE.

There are people though with the same issue that use an all metal hotend and different extruders. Personally i'm using a Revo unit, a Bondtech extruder and a capricorn bowden tube.

ryanklafuric, 2023-04-12:
Strongly agree that this issue is not at all related to the hotend. I ran the test before with a perfectly stock Mini (see above) then I ran the test again with a brand new Slice hotend Copperhead upgrade kit and a SuperPINDA sensor installed and got the exact same results in some very high-quality Atomic PETG (see below).

Print order: Front left > rear left > rear right > front right

Front left:
[image]

Rear left:
[image]

Rear right: 
[image]

Note that the tearing at the seam appears worst here and this is the only cylinder printed **_after_** a left/right printhead travel.

Front right: 
[image]

Upgrades since last test:
- Better filament pathing
- Higher quality filament ([Atomic PETG](https://atomicfilament.com/products/marble-petg-pro?variant=12700911468605&currency=USD&utm_medium=product_sync&utm_source=google&utm_content=sag_organic&utm_campaign=sag_organic&gclid=CjwKCAjwrdmhBhBBEiwA4Hx5gxpU5M4twYEbXUAVGd02QEagAw5r4W3wqVPhsE6_wJZueDcONNj5QBoCquUQAvD_BwE))
- [Slice Engineering all metal Copperhead hotend](https://www.sliceengineering.com/products/prusa-mini-upgrade-kit) installed
- SuperPINDA sensor installed to replace the old PINDA sensor

Notes: I intentionally chose a pretty high quality hotend and roll of filament (brand new, no moisture) to isolate those variables. Given that results are the same (albeit with less stringing) it seems like they aren't factors in the problem.

ryanklafuric, 2023-04-12:
I've got a direct drive kit that I'm going to try installing. Will run this test afterwards to see if that fixes the issue. If that's the case, it would appear that this is related to the bowden tube extruder placement and/or pathing.

rancor1223, 2023-04-12:
After testing my stock Mini+ with 0.6mm ([test results posted before](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2997#issuecomment-1494534114)) I switched it back to 0.4mm. I used the example 3mf file as given. Resulting models are ordered clock-wise starting from bottom right corner of the heat bed.

[image]

I will be switching to Revo soon so I will run the rest again with that.

ryanklafuric, 2023-04-13:
Alright, I just finished the direct-drive conversion and re-ran the test.

**Print order:** Front left > Rear left > Rear right > Front right

Front left:
[image]

Rear left:
[image]

Rear right:
[image]

Front right: 
[image]

Notes:
- All cylinders are basically flawless irrespective of print head travel distance and direction
- The direct drive extruder i'm using is a [Bondtech LGX Lite](https://www.bondtech.se/product/lgx-lite-large-gears-extruder/) with the settings specified by them (0.6a motor current, 35% retraction speed, etc)

As far as I'm concerned, this basically confirms that the issue we're seeing in this bug is caused by the bowden extruder and the bowden tube pathing. An upgraded hotend alone didn't fix the issue for me or others, and adding an upgraded **_bowden_** extruder didn't fix the issue for others either. 

The only thing that fixed this issue for me was the _direct drive_ extruder upgrade.

ScreennameTaken, 2023-04-15:
[image]
Here is a different image to add to my previous ones. Take a look at the feet, the left and the right one. As well as the perimeters.
Thing is, there are no gaps, unless its going from left to right.

irrational314, 2023-04-16:
Figured I'd contribute.

Prusa Mini+ original hotend, Bondtech IFS extruder, firmware 4.40. Fairly recent hotend PTFE replacement. I don't have any Prusament PLA but this is Inland white PLA.

All references to corner will be done from the perspective looking at the printer.
Front right:
[image]

Right Rear:
[image]

Left Rear:
[image]

Front Left:
[image]

abrancat, 2023-04-17:
> Alright, I just finished the direct-drive conversion and re-ran the test.
> 
> **Print order:** Front left > Rear left > Rear right > Front right
> 
> Front left: [image]
> 
> Rear left: [image]
> 
> Rear right: [image]
> 
> Front right: [image]
> 
> Notes:
> 
> * All cylinders are basically flawless irrespective of print head travel distance and direction
> * The direct drive extruder i'm using is a [Bondtech LGX Lite](https://www.bondtech.se/product/lgx-lite-large-gears-extruder/) with the settings specified by them (0.6a motor current, 35% retraction speed, etc)
> * Same filament as before ([Atomic PETG Marble Gray](https://atomicfilament.com/products/marble-petg-pro?variant=12700911468605&currency=USD&utm_medium=product_sync&utm_source=google&utm_content=sag_organic&utm_campaign=sag_organic&gclid=CjwKCAjwrdmhBhBBEiwA4Hx5gwNVCGhNyJJYiLX3xHWN9GDKsoddatejSMFngc4wsWDFCVI1yQK6xBoCE3EQAvD_BwE))
> * Same hotend as before ([Slice Copperhead Upgrade](https://www.sliceengineering.com/products/prusa-mini-upgrade-kit?variant=43063020323057&currency=USD&utm_medium=product_sync&utm_source=google&utm_content=sag_organic&utm_campaign=sag_organic) for the Mini)
> 
> As far as I'm concerned, this basically confirms that the issue we're seeing in this bug is caused by the bowden extruder and the bowden tube pathing. An upgraded hotend alone didn't fix the issue for me or others, and adding an upgraded **_bowden_** extruder didn't fix the issue for others either.
> 
> The only thing that fixed this issue for me was the _direct drive_ extruder upgrade.

Hi all, new to this thread, first time posting on Github.

After 2 failed prints due to a similar underextrusion problems on support structures, I found this thread. I have been experimenting with the files shared by Prusa support as well, and I also believe this has something to do with the X-axis moving left to right with the bowden setup. When it flexes and becomes a sharper angle (left to right) I would bet that it is pulling up some of the filament out of the hotend, which is causing the delayed extrusion after this move. I ran 5 tests with this, only changing the "extra length after restart" in increments - setup and photos below:

Printer type - [MINI]
Mini

Printer firmware version - [e.g. 4.0.5, ...]
4.4.1

Original or Custom firmware - [Original/Custom]
Original

Optional upgrades - [e.g. Filament Runout Sensor]
Filament runout sensor
E3D micro prusa mini edition (but had the same issues with stock hotend)
0.6 mm nozzle

USB drive or USB/Octoprint
Octoprint. But I have the same issue when printing from USB.

Jessie Soul Black Glitter PETG

All images shown in this order: front left, back left, back right, front right

Test 1: Stock Jessie PETG settings in PrusaSlicer 2.6 - 2 issues, back left and back right of print bed
[image]
[image]

Test 2: same settings, plus .3mm extra length on restart - issue only with back right
[image]

Test 3: same, plus .5mm extra length on restart - better, only issue with back right
[image]

Test 4: same, plus .8mm extra length on restart - better yet
[image]

Test 5: This is where I am leaving it for now, I feel it is a good compromise between too much extrusion in most circumstances, but enough to fill the gap in the L>R bowden issue
[image]

Thanks for all the help everyone! Outside of updating the firmware in the printer to account for this, not sure what else can really be done at this point.

ryanklafuric, 2023-04-18:
Hi Michele (@Prusa-Support),

Just wanted to check-in. What's needed from the Prusa side to help fix the under-extrusion issue affecting the Mini? 

Happy to help out some more if needed. Others are too, I'm sure. Just need to know what's needed from our end.

bralexc, 2023-04-23:
Do you need more users to print the test gcode?

3DMusse, 2023-04-25:
I'm so happy that I found this thread. I have had this issue from the start of owning my mini+ and it has driven me insane. It's always on the lower right quadrant of the print bed. I will try the test print as soon as possible to add it to this thread.

rasssta, 2023-04-25:
> Do you need more users to print the test gcode?

Unless you have a print behavior that is different to the ones described above I don't think it's necessary. Thanks anyway! :)

rasssta, 2023-04-25:
> ETA - I will not be sticking with the 1mm extra length on restart for most prints, I believe this will give me far too may globs of filament - but I believe that using the extra length on restart option can be tuned for specific prints to get a consistent compromise.

Yeah, the extra length on restart solves the gaps, but it creates new problems instead. So it's not a proper workaround. I've tried it with different lengths and different objects and you need to fine-tune it for every type of print for it to work decent (and most likely re-print at least once), and that's not OK in the long run.

abrancat, 2023-04-25:
> Yeah, the extra length on restart solves the gaps, but it creates new problems instead. So it's not a proper workaround. I've tried it with different lengths and different objects and you need to [...]

Yeah, I have had to fine tune each print, and for some I have had to paint seams in such a way to avoid having the L=>R movements for a new external seam. Definitely not a workaround, just wanted to share something that helps me for most prints.

ryanklafuric, 2023-04-25:
> Yeah, the extra length on restart solves the gaps, but it creates new problems instead. So it's not a proper workaround. I've tried it with different lengths and different objects and you need to [...]

Strongly agree. I think that this is a fine short-term workaround, but it is finicky and not a sustainable workaround. 

If this was an Ender3, it'd be fine, but given that this printer is [marketed as a workhorse](https://imgur.com/a/ojxxYsB) (and priced like it too) I think Mini/Mini+ users need a proper fix either in the form of a printable hardware update or slicer compensation features.

3DMusse, 2023-04-26:
I, like many others, bought this because it was supposed to be a reliable workhorse (factory assembled). Mine has been nothing but issues. It had random reboots from the very start, even when i printed my first benchy. Figured out that it worked when I used octoprint and skipped the usb-stick entierly. But it took one year to get a new board. 
The issue discussed here on top of that has made it so it mostly collects dust since I can't rely on it actually producing a good print. The positive in all this is that I know a lot about how they work now, but that isn't what my original goal was. I just wanted to get better at 3D-modeling, print my designs and learn the basics. This hasn't been close to the "set it and forget it" experience people have been raving about. I want to sell it, but I see it as defective/broken. An expensive paperweight of frustratiuon and sadness tbh.

Printer type - [MINI]
Mini+

Printer firmware version - [e.g. 4.0.5, ...]
4.4.1+4194

Original or Custom firmware - [Original/Custom]
Original

Optional upgrades - [e.g. Filament Runout Sensor]
0.6 mm nozzle, had the same issue with my 0.4 mm nozzle.

USB drive or USB/Octoprint
Octoprint. But I have the same issue when printing from USB.

Filament: Made for Prusa Sweet Mint PLA+, happens with all my filaments tho.

Don't know if these matter, but I'll add them.

Bootloader Version:
2.0.2

Buddy Board:
1.0.5

--------------------------------------------------------------------------------------

All the pictures are taken standing infront of the printer, the screen on it is facing me.

Front Right:

[image]
[image]
[image]

Front left:
It looks like it printed well, but the seam is so incredibly thin and has some small holes in it. Light just shines through it like nobody's business.

[image]
[image]

Back Right:

[image]
[image]
[image]

Back Left:

[image]
[image]
[image]
[image]

jakemauer, 2023-04-27:
Printer type - [MINI]
Mini+, Factory Assembled

Printer firmware version - [e.g. 4.0.5, ...]
4.4.1+4194
Bootloader 2.0.2
BuddyBoard 1.0.5

Original or Custom firmware - [Original/Custom]
Original

Optional upgrades - [e.g. Filament Runout Sensor]
0.6 mm nozzle

USB drive or USB/Octoprint
Octoprint.

Filament:
Hatchbox White PLA

Here's the PrusaSlicer file that I printed with the modifications to use the HatchBox PLA preset with the other settings from the original prusa file transferred over.
[test1nocrossing-.6nozzle-hatchbox.3mf.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/11346563/test1nocrossing-.6nozzle-hatchbox.3mf.zip)

Photo order: Upper Left, Upper Right, Lower Right, Lower Left
[image]
[image]
[image]
[image]

jakemauer, 2023-04-27:
@Prusa-Support 

With your test prusaslicer file, what attributes should I change when printing with a 0.6mm nozzle? I transferred over all the settings from the file to my .6mm profile but I'm not sure if that's the correct strategy.

jakemauer, 2023-04-27:
Here's another test done with Prusament Galaxy Black

**Printer type - [MINI]**
Mini+, Factory Assembled

**Printer firmware version - [e.g. 4.0.5, ...]**
4.4.1+4194
Bootloader 2.0.2
BuddyBoard 1.0.5

**Original or Custom firmware - [Original/Custom]**
Original

**Optional upgrades - [e.g. Filament Runout Sensor]**
0.6 mm nozzle

**USB drive or USB/Octoprint**
Octoprint.

**Filament:**
Prusament Galaxy Black

Photo order upper left, upper right, lower left, lower right.
[image]

ChrisPirillo, 2023-05-06:
Same printer. Same issue. 0.4mm nozzle. Brand new Prusa assembled hotend. 

[image]
[image]
[image]
[image]
[image]

Prusa-Support, 2023-05-06:
Thank you all.
Probably we have enough information for our developers to carry out a proper investigation.

For now, I agree that _extra length on restart_ may help, **but** it may cause other artifacts (like blobs) where "a delayed extrusion" doesn't occur so it is not very recommended.
All in all, what seems to really help is reducing the _retraction length_, probably at the expense of stringing.

Also, please notice, it doesn't happen frequently to print thin features far from each other. Possibly avoid that as long as you can. 
_Avoid crossing printers_ and _nearest seam position_ may often mitigate the problem by reducing travel, and future confirmed (e.g. experimental [pressure equalizer](https://help.prusa3d.com/article/pressure-equalizer-_331504)) or requested (e.g. https://github.com/prusa3d/PrusaSlicer/issues/7081, https://github.com/prusa3d/PrusaSlicer/issues/8205) features in Prusa Slicer may come in handy too.

In the meantime, our developers devoted to this repository will study the issue from the firmware side.

Michele Moramarco
Prusa Research

ra-kal, 2023-05-08:
I'm fighting with this underextrusion after deretraction since I bought my MINI back in December. It is mainly noticable with PETG. I was never able to solve it and so I gave it to Prusa service for maintenance. It returned back as repaired, but this phenomenon was not solved. Today I ran some tests with different retraction lengs to only notice, that the underextrusion happens only on the right side, eg. after travel from left to right. And while searching, again, for help, I found this thread.

Actually I was searching for another discussion I have seen earlier. Somebody had also such underextrusion problems and was able to solve it by replacing the bowden tube between hotend and extruder with a Capricorn 1.9mm width tube. The real retraction length is probably different on left side, where the tube is stretched to the right side where the tube is bent. Because the Capricorn tube has tighter diameter of 1.9mm only, the difference is probably less and it results in smaller difference in retraction length.

Maybe you can also check it, if replacing the bowden tube would improve it.

[168 more replies]

closed 2025-11-22 by rasssta