prusa3d/Prusa-Firmware-Buddy #3002: Flow Rate/Extrusion Multiplier Function and correction

Fuzarca, 2023-02-23:
Hello,
Stefan from CNC-Kitchen did a lot of experiments measuring the dependence of the real flow rate and the requested flow rate. This depends also on the temperature, filament material and nozzle. That means one should actually work with an extrusion multiplier function instead of a single value. 
It would be awesome to be able to use profiles for each filament and nozzle combinations, where the real flow rate is calculated from the requested one and the temperature. 
That would allow an online correction of the extrusion steps avoiding under and over extrusion at different printing speeds. 
It is definitively an advanced feature and would require previous calibration. Comparable to linear advance. You could also include it in the slicer as part of the filament and printer profile, and do the correction on slicing. 
If there is interest, I am more than happy to elaborate. 

Thank you for your work.
Cheers,

Prusa-Support, 2023-04-05:
Thanks for reaching us.

All of this surely makes sense. I'm afraid the MINI is not equipped with adequate hardware for this kind of advanced correction though.
I have reason to believe that accurate Linear Advance, Extrusion Multiplier, and Max Volumetric Speed, as well as the use of quality filaments, can deliver great print quality.
Further improvements can be expected in the next generation of printers (XL and MK4), which come equipped with a number of new sensors and the possibility to connect an accelerometer. Stay tuned. 🙂 

For now, this issue will remain open as it may be of inspiration for discussions and future developments.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-09-30:
Is it safe to close this issue as obsolete now that `FW 5.1.0-alpha1` is out?

It brings advanced characteristics to Prusa MINIi, such as Input shaper, Pressure Advance, and Precise Stepping that increase flow accuracy and overall print quality.
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0-alpha1

Michele Moramarco
Prusa Research

closed 2023-11-17 by Prusa-Support